arendst/Tasmota #755: Request for adding supla device

roblad, 2017-08-19:
Hi,

Could you Add supla device as supported ?

https://supla.zamel.pl/en/product/row-01/

template

  { "Tas-Supla-R.L.P.B.",   // RL_PB_TASMOTA hardware (ESP8266)

     GPIO_USER,        // GPIO00 D3 Wemos Button Shield
     0,        // GPIO01 TX Serial RXD
     GPIO_USER,        // GPIO02 D4 Wemos DHT Shield
     0,        // GPIO03 RX Serial TXD and Optional sensor
     GPIO_USER,        // GPIO04 D2 Wemos I2C SDA
     GPIO_REL1,        // GPIO05 D1 Wemos I2C SCL 
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_USER,        // GPIO12 D6
     GPIO_REL2,        // GPIO13 D7
     GPIO_USER,        // GPIO14 D5
     0,        // GPIO15 D8
     GPIO_LED1,        // GPIO16 D0 Wemos Wake
     GPIO_ADC0         // ADC0   A0 Analog input
}

arendst, 2017-08-19:
Looking at the schematics (https://www.supla.org/download/wifisocket-schematics.pdf) the connections are totally different from your suggested connections: The relay is connected to GPIO04. A button is connected to GPIO05. A programming jumper is connected to GPIO00 and a three color led is connected to GPIO12-14.

What are the correct connections?

roblad, 2017-08-19:
There is that modyfication, but you can add oryginal supla. My modyfication

https://forum.supla.org/download/file.php?id=171&sid=11d98aeeee6d45114e5655f65dc9d0f1
one error there is , there should bie changed GPIO4 to GPIO5 and back , mistake in schema.

It is device:

shown heare

http://www.wykop.pl/ramka/3325399/diy-supla-do-puszki-instalacyjnej-podtynkowej-supla-org/

PCB avaliable hear

http://dirtypcbs.com/store/user/preview/4069

arendst, 2017-08-19:
Right.

From your schematics I come to this config:
```
  { "Supla Espablo",   // Supla Espablo (ESP8266) - http://www.wykop.pl/ramka/3325399/diy-supla-do-puszki-instalacyjnej-podtynkowej-supla-org/
     0,                // GPIO00 Flash jumper
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_DSB,         // GPIO02 DS18B20 sensor
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_KEY1,        // GPIO04 Button 1
     GPIO_REL1,        // GPIO05 Relay 1 (0 = Off, 1 = On)
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_USER,        // GPIO12 Optional sensor
     GPIO_REL2,        // GPIO13 Relay 2 (0 = Off, 1 = On)
     GPIO_USER,        // GPIO14 Optional sensor
     0,
     GPIO_LED1,        // GPIO16 Led (1 = On, 0 = Off)
     0
  }
```

What input controls relay 2?

roblad, 2017-08-19:
I use for both as switch  - Switch1 and Switch2 

GPIO14 for RELAY1 
GPIO12 for RELAY2

For config I use 

GPIO4 as a Button1

Additionaly there is 
DS18x20 on GPIO2

GPIO0 is set as None

GPIO_LED1,        // GPIO16 Led (1 = On, 0 = Off)

It works according to that Supla by Espablo

Additionaly 

GPIO_ADC0         // ADC0   A0 Analog input

For  LDR  imput 

   +  ------
     3.3V   |
               |
               |
               |
             +-+
             |  \ |  LDR 10kom
             |  \ |
             +-+
                |                    +------\  
                --------------| ADC0  |
                |                    +------/  
             +-+                
             |     |                
             |     | Resistor 10kom 
             +-+                
                |                 
                |                 
                |    Ground             
             -----               
              ---                
               -

arendst, 2017-08-19:
Good luck.

roblad, 2017-08-19:
mmm
On Wiki could you add this device ?
http://www.wykop.pl/ramka/3325399/diy-supla-do-puszki-instalacyjnej-podtynkowej-supla-org/

arendst, 2017-08-19:
You can edit the wiki yourself and in English please.

closed 2018-05-07 by stale[bot]

FialaDavid, 2018-08-11:
If anybody is interested in configuration for inCan v3.3:

`  { "Supla Espablo",   // Supla Espablo (ESP8266) v 3.3
     0,                // GPIO00 Flash jumper
     GPIO_USER,        // GPIO01 Serial RXD
     GPIO_DSB,         // GPIO02 DS18B20 sensor
     GPIO_USER,        // GPIO03 Serial TXD
     GPIO_USER,        // GPIO04 Sensor 1
     GPIO_REL1,        // GPIO05 Relay 1 (0 = Off, 1 = On)
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_KEY2,        // GPIO12 Button 2
     GPIO_REL2,        // GPIO13 Relay 2 (0 = Off, 1 = On)
     GPIO_KEY1,        // GPIO14 Button 1
     0,                // GPIO15 Permanent LOW
     GPIO_USER,        // GPIO16 Sensor 2
     GPIO_ADC0         // ADC0   A0 Analog input
  },`