arendst/Tasmota #756: Upload issues

efp1, 2017-08-19:
I'm having issues uploading the firmware to my sonoff switch. I've tried using both platformio ide and arduino. They both start uploading fine, but get stuck when they reach over 85% uploaded. It doesn't give me an error, it just remains stuck at uploading.

Any ideas? I've tried searching for similar issues but can't find any. 

Thanks!

efp1, 2017-08-19:
[image]

Here's a screenshot as an example. I've tried it several times and it keeps happening.

efp1, 2017-08-24:
No ideas as to what can be causing it?

pierrerigal, 2017-10-07:
@efp1 same problem here, did you find any solution ?

pierrerigal, 2017-10-07:
Solved. Don't forget to "Clean" between 2 uploads.  
Platform.io > Clean

ol-v-er, 2017-10-23:
I confirm. Cleaning the project before a new upload solve the problem

closed 2018-05-07 by stale[bot]