arendst/Tasmota #758: The lights go out and go on without warning

albukrek, 2017-08-19:
**First of all Thanks, for your wonderful work!**

(All devices have Sonoff-Tasmota-5.5.0)

I have installed  three Sonoff units a few days ago. 
- (2) **Sonoff basic**
     - this units are  responsible for the night lamps
- (1) **Sonoff 4-channel** 
    - One channel is responsible for the main light in the room
    - The other three channels control the fan.

**All the lamps** are connected on one side to the old switches on the wall, and on the other side to GPIO  14 in the basics,  and the 4-channel lamp switch to GPIO 3.

The old fan switch has been replaced by wemos d1 pro so that the Sonoff GPIO is not used for the fan channels.

In general, everything works properly, but when you turn on or off (this happened only with the lights) one of the lamps manually, occasionally without warning the lamp toggles.

We think it happens only in case that the old switch is used, so in order to test this hypothesis yesterday and today we lit the light only through the Home Assistant and the toggling have  disappeared, except for one case of shutdown and fire back almost immediately.

I'm sure that I have done something wrong, please advise me how to fix this problem.

Thank's
Arik

albukrek, 2017-08-27:
I Think that there is a simple solution for my problem:

1. The Sonoff devices  were updated to version5.6.1
2. Change the topic of the mqtt from stat to cmnd (for more detailes see this wiki [topic](https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic#switchtopic) ) 
Run commands: 
SwitchTopic 2.
SwitchMode to 1
I'm not sure, but maybe this way it the switch state is retained in the MQTT

albukrek, 2017-08-28:
The above fix has one drawback, every 50 minutes or so the relay flickers.

Can anyone help please help me?

albukrek, 2017-08-28:
Found the problems:

1. Topic name in home assistant was written in lower case, while in Sonoff-Tasmota the name is all in upper case.
2. Some retain topic were left on the MQTT server, these topics were deleted.

Thanks!
[image]

PeggyFree, 2017-08-29:
How do you check such retained messages ? My mosquito mqtt server is
running as a service, I could'nt find any solution to check all the mqtt
trafic and the potential residual subscriptions.

davidelang, 2017-08-29:
see the wiki page on poweronstate, it tells you how to check for retained 
messages

mkh595, 2017-08-29:
You can see retained messages e.g. in MQTTfx client. If you subscribe topic, you will receive first all retained messages (if there are any)

[image]

albukrek, 2017-09-03:
Sorry to report, but the issue have returned...
This time it only happens when the switch is in an open state (circuit is open) from #665 and others 
The solution may be adding a resistor and capacitor

albukrek, 2017-09-08:
Working fine after adding a pullup resistor and capacitor.

closed 2017-09-08 by albukrek

aspitznyc, 2017-10-23:
Hi @albukrek what size of capacitor did you use? And where did you put it? Between which pins? Thanks!!!

davidelang, 2017-10-23:
anything in the pF range should be ok.

aspitznyc, 2017-10-24:
Sorry about my lack of electronics understanding, but could you provide and example? I'm a noob in all this.

Thanks for your help!

davidelang, 2017-10-24:
see https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs

On Tue, 24 Oct 2017, aspitznyc wrote:

> Date: Tue, 24 Oct 2017 04:41:14 -0700
> From: aspitznyc <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f0705e7c71d1d6e1cf2f0f76f7933e7d255b510dc92cf000000011606e8d
>     a92a169ce0efcd1c7@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] The lights are flickering without
>     warning (#758)
> 
> Sorry about my lack of electronics understanding, but could you provide and example? I'm a noob in all this.
>
> Thanks for your help!
>
>
>