arendst/Tasmota #760: MQTT - Wrong topic in response to a command topic - 5.3.0

s89511, 2017-08-19:
I'm using 5.3.0 release.

I configured my sonoff basic with 
topic = **sonoffbasic01**
fulltopic = **tasmota/garden/light/%topic%/%prefix%/**

When I send the MQTT message **tasmota/garden/light/sonoffbasic01/cmnd/POWER ON** the device reply with 
**stat/sonoffbasic01/RESULT = {"POWER":"ON"}**
**stat/sonoffbasic01/POWER = ON**

Is this the expected behaviour? It used the old topic format and not the configured fulltopic.

sillyfrog, 2017-08-20:
I think I'm running into a similar issue in v5.6.1b (on a Sonoff Basic, rebuilt on the latest version just now).

When sending the command:
`tasmota/sonoffbasic1/cmnd/Power1 TOGGLE`

The response I receive is:
`stat/sonoffbasic1/POWER OFF`

However, if I press the Button on the Sonoff Basic (or flip the switch connected to GPIO14), I get a the MQTT message:
`tasmota/sonoffbasic1/stat/POWER OFF`

arendst, 2017-08-20:
Pls provide output of commmand status 0.

I suspect your topic is equal to fallbacktopic = mqttclient.

sillyfrog, 2017-08-20:
Thanks!

I think I'm missing something as I don't see a fallbacktopic (below or in the docs) :/

Here is the output of both `status0` and `status 0` (slightly sanitised).

```
tasmota/sonoffbasic1/cmnd/status0 (null)
stat/sonoffbasic1/STATUS {"Status":{"Module":1, "FriendlyName":"SonoffBasic1", "Topic":"sonoffbasic1", "ButtonTopic":"0", "Power":1, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
```

```
tasmota/sonoffbasic1/cmnd/status 0
stat/sonoffbasic1/STATUS {"Status":{"Module":1, "FriendlyName":"SonoffBasic1", "Topic":"sonoffbasic1", "ButtonTopic":"0", "Power":1, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
stat/sonoffbasic1/STATUS1 {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":1, "Sleep":50, "BootCount":21, "SaveCount":224, "SaveAddress":"F8000"}}
stat/sonoffbasic1/STATUS2 {"StatusFWR":{"Program":"5.6.1b", "BuildDateTime":"2017-08-20T12:56:35", "Boot":4, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoffbasic1/STATUS3 {"StatusLOG":{"Seriallog":0, "Weblog":2, "Syslog":0, "LogHost":"domus1", "SSId1":"wifi", "SSId2":"indebuurt2", "TelePeriod":300, "Option":"55800009"}}
stat/sonoffbasic1/STATUS4 {"StatusMEM":{"ProgramSize":477, "Free":524, "Heap":26, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}}
stat/sonoffbasic1/STATUS5 {"StatusNET":{"Hostname":"sonoffbasic1-4333", "IPaddress":"10.7.10.202", "Gateway":"10.7.255.254", "Subnetmask":"255.255.0.0", "DNSServer":"10.7.255.254", "Mac":"5C:CF:7F:1C:30:ED", "Webserver":2, "WifiConfig":2}}
stat/sonoffbasic1/STATUS6 {"StatusMQT":{"Host":"openhab.sf", "Port":1883, "ClientMask":"sonoffbasic1", "Client":"sonoffbasic1", "User":"DVES_USER", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
stat/sonoffbasic1/STATUS7 {"StatusTIM":{"UTC":"Sun Aug 20 07:12:34 2017", "Local":"Sun Aug 20 08:12:34 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
stat/sonoffbasic1/STATUS10 {"StatusSNS":{"Time":"2017-08-20T08:12:34", "Switch1":"OFF"}}
stat/sonoffbasic1/STATUS11 {"StatusSTS":{"Time":"2017-08-20T08:12:34", "Uptime":1, "Vcc":3.231, "POWER":"ON", "Wifi":{"AP":1, "SSID":"wifi", "RSSI":100, "APMac":"08:EA:44:67:F9:D7"}}}
```

arendst, 2017-08-20:
As said fallbacktopic = mqqtclient name.

Your mqttclient name is "Sonoffbasic1". Your topic name is equal. This results in your experienced behavior because for Tasmota they need to be different. 

- The mqttclient name needs to be unique for MQTT as the client.
- The mqtt topic may be unique too to address a specific device.
- Tasmota subscribes to both topic name AND mqttclient name (as fallback topic name in case you messed up with your topic name). The fallback topic name result message is always stat/mqttclientname/result. 
- The topic name result message uses the fulltopic structure only.

SOLUTION: Make your mqttclient name unique AND different from topic name.

sillyfrog, 2017-08-20:
That sorted it, thanks!

Changing the `Client Id` to be different from the `Topic` means the generated `FallbackTopic` will now be different. If possible, it would be great to update the Web UI with a hint to keep the two fields different (as I left the rest as the default) for us newbies :) 

Thanks again for you prompt help, really appreciate it!

s89511, 2017-08-20:
Thanks. I changed MqttClient and now it works as expected.

This is great 👍 

Thank you very much for your support.

closed 2017-08-20 by arendst

nsitt, 2021-01-13:
Getting the same issue but Client, Topic and Fallbacktopic are distinct so the fix mentioned in this thread does not work.

I have multiple tasmota devices configured and I am unable to send a MQTT command to the tasmota devices from one to another. 
The response is coming back with information from the current devices and not from the remote device.

As an example, here is a mqtt command from "display" to "D1miniUG".

I copied console output from both devices, 

1. Display

###############
16:23:50 CMD: cmnd/D1miniUG/Power
****16:23:50 MQT: stat/Display/RESULT = {"POWER":"OFF"}**
16:23:50 MQT: stat/Display/POWER = OFF**
16:23:53 CMD: status 0
16:23:53 MQT: stat/Display/STATUS = {"Status":{"Module":0,"DeviceName":"Display","FriendlyName":["Display"],"Topic":"Display","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:23:53 MQT: stat/Display/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:07:51","StartupUTC":"2021-01-13T15:16:02","Sleep":50,"CfgHolder":4617,"BootCount":24,"BCResetTime":"2021-01-11T16:12:18","SaveCount":78,"SaveAddress":"3F5000"}}
16:23:53 MQT: stat/Display/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2021-01-11T16:10:49","Boot":6,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"382/699"}}
16:23:53 MQT: stat/Display/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["athome",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
16:23:53 MQT: stat/Display/STATUS4 = {"StatusMEM":{"ProgramSize":611,"Free":2460,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FA0478E","04361FA1","000600C3","00000000","C000F981","00000024","00001000"],"Drivers":"1,2,4,6,8,9,10,13,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,4,5,6,14,15"}}
16:23:53 MQT: stat/Display/STATUS5 = {"StatusNET":{"Hostname":"Display-8154","IPAddress":"192.168.178.104","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"48:3F:DA:00:9F:DA","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:23:53 MQT: stat/Display/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.5","MqttPort":1883,"MqttClientMask":"Display","MqttClient":"Display","MqttUser":"<name>","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:23:53 MQT: stat/Display/STATUS7 = {"StatusTIM":{"UTC":"2021-01-13T15:23:53","Local":"2021-01-13T16:23:53","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:39","Sunset":"17:18"}}
16:23:53 MQT: stat/Display/STATUS10 = {"StatusSNS":{"Time":"2021-01-13T16:23:53","SHT3X-0x44":{"Temperature":21.6,"Humidity":52.3,"DewPoint":11.4},"MHZ19B":{"Model":"B","CarbonDioxide":895,"Temperature":21.0},"TempUnit":"C"}}
16:23:53 MQT: stat/Display/STATUS11 = {"StatusSTS":{"Time":"2021-01-13T16:23:53","Uptime":"0T00:07:51","UptimeSec":471,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"athome","BSSId":"88:D7:F6:B6:84:90","Channel":8,"RSSI":30,"Signal":-85,"LinkCount":1,"Downtime":"0T00:00:03"}}}

############
2. D1miniUG

16:21:52 CMD: status 0
16:21:52 MQT: stat/D1miniUG/STATUS = {"Status":{"Module":0,"DeviceName":"D1_mini_UG","FriendlyName":["D1_mini_UG_FN"],"Topic":"D1miniUG","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:21:52 MQT: stat/D1miniUG/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T20:36:31","StartupUTC":"2021-01-12T18:45:21","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2021-01-09T12:28:01","SaveCount":31,"SaveAddress":"F5000"}}
16:21:52 MQT: stat/D1miniUG/STATUS2 = {"StatusFWR":{"Version":"9.1.0(sensors)","BuildDateTime":"2020-11-07T11:59:31","Boot":6,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"403/699"}}
16:21:52 MQT: stat/D1miniUG/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["athome",""],"TelePeriod":500,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
16:21:52 MQT: stat/D1miniUG/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
[... cut ...]
16:21:52 MQT: stat/D1miniUG/STATUS10 = {"StatusSNS":{"Time":"2021-01-13T16:21:52","SHT3X-0x44":{"Temperature":21.1,"Humidity":44.4,"DewPoint":8.5},"MHZ19B":{"Model":"B","CarbonDioxide":478,"Temperature":23.0},"TempUnit":"C"}}
16:21:52 MQT: stat/D1miniUG/STATUS11 = {"StatusSTS":{"Time":"2021-01-13T16:21:52","Uptime":"0T20:36:31","UptimeSec":74191,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"athome","BSSId":"88:D7:F6:B6:84:90","Channel":8,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}