arendst/Tasmota #761: Sonoff T1 Touch Switch Relay goes high on boot

crgor, 2017-08-20:
Hi All,

I have been playing with the Sonoff T1 1CH/2CH/3CH touch switches, I have noticed an issue with Relay going high on boot (regardless of the state it was before power down) and resetting back to the last state it was. 

Has anyone noticed this issue with any other models of Sonoff? How do I fix this?

Any help would be appreciated.

Cheers,
Reddy.

arendst, 2017-08-20:
I noticed it too on the T1. I'm afraid the hardware has been setup that way. I'm unable to get logics of the device but I'm afraid you'll have to live with it.

Just tested the Touch and that one seems OK.

crgor, 2017-08-21:
Thanks for confirming.

bortek, 2017-09-04:
I am planning to buy several of T1. Just want to clarify the behavior. When booting T1 it will go on is that correct? Then you just switch it off either by pressing touch button or using remote command and then it works just normal, is that correct?

crgor, 2017-09-05:
Yes that is correct, another way to switch it off is write your firmware
such a way that during booting you set the state of the relay as low (off),
then set what ever the last saved state.

asantos15, 2018-02-26:
Hello there
It's been happening to mine, too, but I got an inching/self locking models that does the same and that would be enough to open a gate...so it gets unusable for that purpose.