arendst/Tasmota #762: Connect failed as AP cannot be reached

omarusman, 2017-08-20:
Hi guys, can you help me. I'm trying to upload v5.6.1  but I couldn't connect to WiFi.

These what's in my serial log at LOG_LEVEL_DEBUG. Am I missing something?
```
00:00:00 Cnfg: Use defaults
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_E9DBD3, GroupTopic sonoffs) Version 5.6.1
00:00:00 Cnfg: Save (1648 bytes) to flash at FA 00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Attempting connection...
00:00:04 Wifi: Attempting connection...
00:00:05 Wifi: Attempting connection...
00:00:06 Wifi: Attempting connection...
00:00:07 Wifi: Connect failed as AP cannot be reached
00:00:07 Wifi: Patch issue 2186
00:00:07 Wifi: Connecting to AP2 amigu in mode 11N as sonoff-7123...
00:00:08 Cnfg: Save (1648 bytes) to flash at F9 and count 3
00:00:08 Wifi: Attempting connection...
00:00:09 Wifi: Attempting connection...
00:00:10 Wifi: Attempting connection...
00:00:11 Wifi: Attempting connection...
00:00:12 Wifi: Attempting connection...
00:00:13 Wifi: Attempting connection...
00:00:14 Wifi: Connect failed as AP cannot be reached
00:00:14 WPSconfig: Active for 1 minute
00:00:29 WPSconfig: FAILED with status 4
00:00:33 APP: Restarting
```

omarusman, 2017-08-22:
I've tried installing **v5.0.+** and it's working fine.  can connect to WiFi.

So I was curious if **v5.6.+** is a problem, so I did OTA firmware and it was uploaded successfully.

But when I run the device, I get the same issue above.

mkh595, 2017-08-24:
I had same problem yesterday upgrading 3 POW's to 5.6.1c. Both ways, new flash or OTA. I had to several times start WiFi config, re-scan, select WiFi to established connection after upgrade.

closed 2018-05-07 by stale[bot]