arendst/Tasmota #763: Adding 2 things in the web - new config - request

roblad, 2017-08-20:
Hi,

One more thing if I can ask you :-)

In the MAIN MENU>CONFIGURATION>CONFIGURE MODULE

behind of i.e GPIO12 there is description, but it is hard-coded in the example there is Relay1.
Is it possible to make that configurable as edit box where anybody will be able to describe GPIO function specyfic for him.

Second thing, is it posible to prepare description of buttons i.e change Togle1 Togle2 to my own description i.e. first button as "Lamp top" and second "Lamp left" etc. some config check box is nice to have. 

Additionally when you could predict one more file with variables where will be configured language pack  i.e. I can translate then for Polish language and so on.  I.e. as it is there

https://github.com/obsy/gargoyle/blob/master/package/gargoyle-i18n/JS%20Translation%20Guidelines%20(for%20developers).txt

It will be very useful for ordinary users without huge expirience of IT, when somebody will prepare device for i.e. family for easy usage and without english speaking/reading  :-)

Thanks,

arendst, 2017-08-22:
Working on i18n....

arendst, 2017-10-30:
Implemented internationalization starting with version 5.7.0

closed 2017-10-30 by arendst