arendst/Tasmota #764: cannot compile

andreasloe, 2017-08-20:
I donloaded the latest code (version 5.6.1.c) today but cannot compile it. The problem has to do with the IRsend-command. I am getting the following error message

`drv_ir_send:48: error: 'IRsend' does not name a type`
` IRsend *irsend = NULL;`
` ^`
`/Users/andies/Documents/Arduino/libraries/Sonoff-Tasmota-development/sonoff/xdrv_ir_send.ino: In function 'void ir_send_init()':`
`xdrv_ir_send:52: error: 'irsend' was not declared in this scope`
`   irsend = new IRsend(pin[GPIO_IRSEND]);  // an IR led is at GPIO_IRSEND`
`   ^`
`xdrv_ir_send:52: error: expected type-specifier before 'IRsend'`
`   irsend = new IRsend(pin[GPIO_IRSEND]);  // an IR led is at GPIO_IRSEND`
`                ^xdrv`

I  remember last time that I added 
` #include <IRsend.h>  // Needed if you want to send IR commands.`
` #include <IRrecv.h>  // Needed if you want to receive IR commands.`
at line 27/28 in xdrv_ir_send.ino and that solved my problem. But this time I was not successful, the error message is attached since it is so long.

[error.txt](https://github.com/arendst/Sonoff-Tasmota/files/1237246/error.txt)

andreasloe, 2017-08-20:
After including the two lines I am getting this message

`/Users/andies/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /var/folders/l6/myrckv8d12z9_06fl49fm36h0000gn/T/arduino_build_803675/sonoff.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status`

davidelang, 2017-08-20:
It sounds as if you have not gone through the wiki prereq section, so you are 
missing a library (and probably others)

andreasloe, 2017-08-20:
I went throught that section but was not able to install it via the library manager, see screenshot below (no version 1.0.2 available). Therefore, I downloaded it by hand and I used the newest version. Is that the problem?
[image]

davidelang, 2017-08-20:
This error normally means that it didn't find the library, do a more verbose 
compile and I'll bet that it's not finding it.

andreasloe, 2017-08-20:
But doesn't
`Bibliothek IRremoteESP8266 in Version 2.1.0 im Ordner: /Users/andies/Documents/Arduino/libraries/IRremoteESP8266  wird verwendet`
say that it is using this library? (The error message appears earlier.)

andreasloe, 2017-08-20:
I remember now what I did and what solved my problem. I added the lines 25 and 26 in the screenshot below.

[image]

But who's problem is this? Is my installation corrupt?

arendst, 2017-08-20:
As the wiki tries to explain in several cases the only supported libraries are provided in the repository and the release Source code download.

closed 2017-08-21 by andreasloe