arendst/Tasmota #766: Sonoff 4CH very slow

paulo9999, 2017-08-21:
Hello, I've just installed the 5.5.1 version in Sonoff 4CH. It works ok, but the response is very very slow when I test it at the webpage. When I click at toggle, at the first time the response is fast, but by the time it takes 2 to 5 seconds to respond.

I tried to send the commands using the console, The same thing happens. It takes up to 5 seconds to display the command and to execute it.

Is that normal?

arendst, 2017-08-21:
Check your mqtt connection using weblog 3.

aaamitsingh, 2018-04-15:
@arendst, I have Sonoff 4 ch and Sonoff Pow all are working fine but I'm facing the similar issue with Sonoff-4ch-Pro. 1st-time both button and MQTT command are working fine but after that, it's responding very late and sometimes not responding. Is it a hardware issue?