arendst/Tasmota #768: External wall switch with sonoff basic?

habanero79, 2017-08-21:
I bought a sonoff basic switch for the first step to home automation. I see the video "sonoff secrets" on YouTube where the actor connects an usual wall switch to gpio 14 and ground to toggle the light.

Now I have the sonoff and tried this. But nothing happens. After this I read that sonoff disabled this feature in the newer firmware.

The thing is, in my home there are usual wall switches everywhere and I couldn't spare this because of my wife who doesn't like home automation ;). So I definitely need common switches on the wall.

Is there a way to realise this? At the moment I have two sonoff basic with original software. My skill in electrical work is not like a pro but more than basic. So I can handle a solder stick. I live in Europe and in the wall there is L,N and PE.

Thanks

sillyfrog, 2017-08-23:
Sonoff-Tasmota allows you to do exactly this. I have done the same thing using GPIO 14.

To update the flash, I have packaged up my work with the SonOTA project here so it's quick to get going: https://github.com/sillyfrog/SonOTA

Once flashed, you can then configure your Sonoff via the new Web UI to change the module to have "GPIO14" as "Switch1", and it will do what you need (when ever the switch is changed, the light will also change).

You will then need to configure it for what ever home automation you have (eg: using MQTT), but there are many guides on this once you have it up and running.

Good luck!

habanero79, 2017-08-23:
Thank you. After flashing so I need a server eg on a raspberry to get the sonoff working or is the switch still workalone?

I know, the app doesn't work anymore and I have to use the website of the switch. But can I do this without a server?

Maybe later I will configure a raspberry as server.

sillyfrog, 2017-08-23:
If you want to turn it on or off via the Web UI, you can do this without a server from any web browser on your network.

titomontes, 2017-08-29:
@sillyfrog: thanks for your support. I´m having the same issue; I´ve tried setting switchtopic from 1 to 4 and selecting GPIO14 as "09 Switch1" (strangely, next to GPIO14 it says "Sensor" in the heading), but none of those combinations have worked.
I´ve researched enough to kwnow that GPIO14 has it´s own pull up resistor, so that can´t be it...

davidelang, 2017-08-30:
On Tue, 29 Aug 2017, titomontes wrote:

> I´ve researched enough to kwnow that GPIO14 has it´s own pull up resistor, so that can´t be it...

for several people adding an external pull-up resistor to GPIO14 has solved 
their problems.

for several more they've needed to add a bypass capacitor to GPIO14

PeggyFree, 2017-08-30:
Considering this, I think it is easier to hack the original push button.
I'll do this for my next hack. Can we turn it to a switch button, like for
the gpio14?

sillyfrog, 2017-08-30:
I'm assuming this is a Sonoff Basic? When you say SwitchTopic set to 4, from my understanding, that will mean the switch will start to just send MQTT messages to the MQTT topic "4" - probably not what you want? I would suggest setting this to "0" again, and setting "SwitchMode1" to "0".

In the web interface I have selected:

- GPIO14 Sensor
- 09 Switch 1

Then my external toggle switch connected to the bottom 2 pins (GND and GPIO14) of the relevant header as per here: https://github.com/arendst/Sonoff-Tasmota/wiki/GPIO-Locations

titomontes, 2017-08-30:
FINALLY!

Working as expected!, Many thanks!

habanero79, 2017-08-30:
Congrats!

Could you please explain what exactly you have done?

titomontes, 2017-09-03:
For closure: 
- Switchtopic set to ´Sonoff1´ (or whatever topic you want to publish to
- Switchmode1 set to ´0´
- Wallswitch connected to board headers 4 and 5[ (GND and GPIO14)](https://github.com/arendst/Sonoff-Tasmota/wiki/GPIO-Locations). 
- In the webserver/configuration/configure module: select GPIO14 Sensor, set it to: 09 Switch 1

And voilá!

habanero79, 2017-09-03:
Thanks a lot. I will try it. But I have a new problem...

I life in Germany where there is not much space behind the wall switch. I can't put the sonoff in there. Maybe without the housing but I'm not sure if I want to do this because of the open 220volt... So i think the better way is to put the sonoff on the lamp side. But I'm not sure how to connect the wall switch.

In the wall I have phase, neutral and ground. The switch switches phase. Neutral is always connected. So how can I put the sonoff on the lamp side with these cables... I can always connect the phase and neutral. So the sonoff has power. But how to connect the switch? I have the "unused" ground cable. Can I use the ground and the neutral cable to the wall switch and the other side to sonoff (ground and gpio14)? Does the the neutral cable destroy the sonoff board if it is connected to the ground or gpio14 pin?

I know it's not a good solution but I don't know how to connect it all without hammering of the wall...

titomontes, 2017-09-04:
I´m planning on putting the sonoff inside the wall. Yes, the sonoff needs 220V current, so it needs two leads, although it only switches one cable. As long as there is no power on the neutral/GND cables, you can assign them to the wall switch and connect them to GPIO14/GND. GPIO14 is a sensor lead, so it won´t admit current. FYI, the neutral cable also has current on it (try connecting it along with gnd to anything, you´ll get power)

Cheers

habanero79, 2017-09-05:
Okay. I see it is not a good solution. Maybe I can try to use a Relais for this? The connections were:

Phase and neutral to sonoff, so it has power. The other line, normally used as ground, will be used for the wall switch to switch another phase connection. On the lamp side I will use a normal Relais which is connected to the neutral and the phase from the switch. Like this one (FIN-55-34-8)

If the wall switch turns on, the Relais also turns on and makes a second bridge separated to the 220v power circle (is this the correct word in English?!). This bridge, who doesn't carries any power I can use for the gpio14/gnd connection.

Does this work?

mkh595, 2017-09-05:
`GPIO14 is a sensor lead, so it won´t admit current.`

GPIO14 is Input/Output, it depends how do you configure it. Set is as Relay, Led or Pwm and current will be joyfully flowing.

ramgopal5, 2018-01-27:
This is exactly what I want to do. I was wondering if I can still use the original sonoff firmware and instead of using GPIO14 what if I just solder two wires to the existing push button switch of sonoff. So left foot goes to live wire and right foot of the push button is soldered to black cable going to lights.
Will that work ?. 
I don’t want to flash the sonoff yet cause I can’t go back and then I might have to revert to other solutions like homeautomations.

vikasbhasin, 2018-03-25:
Hi All, I'm also new to automation and trying to do the same as OP however, I'm thinking of using ESPurna rather Tadapta. Has anyone achieved same results with ESPurna firmware?

dangyuluo, 2018-04-06:
@sillyfrog Hi. I'm thinking of get use of the GPIO 14. Do you know if it has pull-up resistor inside? Not very familiar withe the circuit.

vaz83, 2018-07-05:
Hi everyone! I have recently flashed a sonoff to add a external switch and added a 10k resistor from GPIO14 to +3.3v, and a 100pF capacitor between GPIO14 and GND. the length of the wires to the switch is about 1,5 meters, and from time to time i still get the light turning on and of by itself. Can anyone help me out on what should i do more? I have read everything i could regarding this issue but haven't found a solution yet.

marcelobaez, 2018-08-01:
@titomontes did you ever had any reboots or blinkings with your solution?

vaz83, 2018-08-01:
I have blinkings

dangyuluo, 2018-08-01:
No there is not. Maybe you need to add a capacitor.

Best,

--------
Hao Peng
Email: hao@pengs.us

vaz83, 2018-08-01:
I did.. 100pF

marcelobaez, 2018-08-01:
> This is exactly what I want to do. I was wondering if I can still use the original sonoff firmware and instead of using GPIO14 what if I just solder two wires to the existing push button switch of [...]

No, there's no way to achieve this on the original firmware. You need to flash with tasmota (or other) and configure it to do so

marcelobaez, 2018-08-01:
> I have blinkings

Did you configured like this? 

> In webserver/console:
> Switchtopic set to Sonoff1 (or whatever topic you want to publish to)
> Switchmode1 set to 0
> In the webserver/configuration/configure module: select GPIO14 Sensor, set it to: 09 Switch 1
> Wallswitch connected to board headers 4 and 5 (GND and GPIO14).

neeravmakwana, 2020-06-02:
> Thanks a lot. I will try it. But I have a new problem...
>
> I life in Germany where there is not much space behind the wall switch. I can't put the sonoff in there. Maybe without the housing but I'm [...]

I am trying to do this too. I have flashed my Sonoff basic with Tasmota and set GPIO14 to Switch1 and tried 2 variants of SwitchMode - 0 and 1.

My issue is related to wiring. I have replaced switches, lights etc. but am not well versed with many electrical concepts. I thought I could connect my Sonoff to the extranal wall switch on the lamp side by doing the following:
1. Open the lamp
2. Cut the L and the N wires
3. Connect the L and N wires coming from the wall to L and N of Inputs of the Sonoff
4. Connect the L and N wires going to the lamp to L and N of Outputs of the Sonoff

I have not done any other soldering or connection (some people mentioned connecting GPIO14 to Ground...) on my Sonoff.

When the power switch is ON, I can connect the device using the web interface, MQTT etc. But when I turn the wall switch off, the power to the Sonoff also gets cutoff and it no longer has internet connection.

I am sure this is expected because and I am doing something wrong on the wiring side. But I don't know how to fix it. Can someone please advise. Thanks.