arendst/Tasmota #769: Sonoff G1 GSM GPRS Module

EmilySmith0, 2017-08-21:
@arendst  
Can you please add code for Sonoff G1 GSM/GPRS.
http://sonoff.itead.cc/en/products/sonoff/sonoff-g1
Looking Forward
Thanks 
Emily

ps2mdg, 2017-10-20:
Yes, please i need code for g1

karobyy, 2017-10-24:
I live in serbia. In my country have 3 gsm operator, and only work with 1 (VIP telekom)

AHBkingman, 2018-03-13:
Sonoff G1 will improve M2M IoT a lot. Please if somebody had developed a firmware for G1 please upload it.

mrrcomp, 2018-04-04:
Hi
Does anyone have info on the G1 to allow it to Transmit Sensor info to mqtt??

eazuka, 2018-05-16:
@arendst any hope of Sonoff G1 being supported

closed 2018-07-15 by stale[bot]

joe63, 2018-07-25:
@arendst any hope of Sonoff G1 being supported, it been a wish come true for my use case, now remaining Tasmota support

vshaev, 2018-08-10:
@arendst any hope of Sonoff G1 being supported, it been a wish come true for my use case, now remaining Tasmota support

i3laze, 2018-11-13:
Up.

obilginer, 2020-04-06:
is there any news?

AdrielGH, 2021-08-29:
Hello all, news?