arendst/Tasmota #771: Flash/Upload issues, which FTDI is best.

ramoun, 2017-08-23:
Hi,

I'm having flashing issues ( newer sonoff recent purchase), I've tried PlatformIO and Arduino, seperate power supplies and cables, different sonoff rf (basic) units, USB cables, bread boards, jumpers, even different FTDI modules from 2 different manufacturers.  I'm using Win7 (64)

Noting there are a lot of recent posts with flashing problems for newer sonoffs.

A couple of times it looked like it worked using platformIO and these rough steps: 
- disconnect FTDI from sonoff and USB port.
- connect FTDI to USB
- start upload/flash (platformIO)
- connect FTDI to sonoff while holding button, release just before upload starts

Which leads me to guess it is a timing issue related to the FTDI, and possibly the quality of the FTDI

Which FTDI is everyone using, which is best, chip sets etc.

mkh595, 2017-08-23:
There are many videos on youtube how to bring Sonoff to flash mode. Your steps are not correct

- connect everything
- press and hold button
- disconnect +VDD from Sonoff for a while
- connect +VDD back
- release button
- LED should blink
- start upload in platformio

ramoun, 2017-08-23:
Thanks, I've watched many to ensure I was not missing any tricks, getting the ESP8266EX into flash mode should be easy. 

Interestingly using the steps above, I dont get the LED blink.  If the upload continues I see the common error:
 warning: espcomm_sync failed
 error: espcomm_open failed
 error: espcomm_upload_mem failed

giig982, 2017-08-23:
Try to never release the button until upload finishes

mkh595, 2017-08-23:
What do you mean "newer sonoff" ?
If it is new not yet flashed and you connect FTDI then LED should blink. If you switch to flash mode, LED will blink quickly. If it is not blinking then you are not in flash mode. Chip can be broken.

What do you mean "If the upload continues" ?
Was upload started showing progress (.....) and in the middle failed? Or didn't started?

If you have already some sonoff with TASMOTA firmware, just connect it to FTDI and open serial monitor in platformio. You should see log. If you cannot see log then cable, FTDI or usb have issue.

ramoun, 2017-08-24:
Thanks for the help/comments.

"Newer sonoff" Unit is sonoff RF, HW QC stamp has 15.5.2017, on the underside of the board roughly in the center it has an odd capacitor that appears to be hand soldered/bridged diagonally between D5 and C?.  The 4 sonoff rf units I've opened all have this same mod.

"If upload continues" It does not reach the (......... )

On the one unit that did flash I have comms as I can see the unit trying to connect to the wifi etc.

Using a fresh unit from the box, connected to FTDI the unit flashes at a ratio of about 3:1, 3 pause - 1 flash. Pushing the button toggles the relay.

If I remove the plug (pwr,gnd,rx,tx) from the sonoff board while holding the button, wait a few sec then reconnect, the LED does not light. Its the same if I remove the plug, wait, hold button, then connect again.

Even the unit that did flash does not blink after trying to put it in flash mode.  

While there is always a chance of a faulty ship you would not think all 3 units would be faulty unless there was a bad batch, if this was the case I would have thought sonoff would have had trouble flashing them.

I also tried a TH10 POW has V2.0 on the board, it behaves the same way

ramoun, 2017-08-24:
Tested sonoffs with esptool.exe and ESPEasy, It would appear that 4 of the FTDI are faulty, 2 can upload but fail at varying points, and the other 2 wont upload while they receive ok.
They all use FT232RL ship, I've ordered a couple with CH304G and CPL2102 chip as they are apparently better.
I came across this article out of interest http://www.usconverters.com/index.php?main_page=page&id=62

I dont get any LED flashes when it goes into flash mode and uploads successfully.

closed 2017-08-24 by ramoun