arendst/Tasmota #773: Button usage in Sonoff Touch

Sunrise17, 2017-08-23:
Dear @arendst

Regarding to following option below in Button usage;
"Pressing the button for over four seconds: Reset settings to defaults as defined in user_config.h and restarts the device"

This is causing an issue in Sonoff touch when people who are not aware of these products, using Sonoff Touch with long press to switch on lights so it causes Sonoff touch to reset user_config.h settings which is not consisting my settings. Because my settings are locating in user_config_override.h. 
I just want to ask you that Is it possible to disable "Pressing the button for over four seconds" activity from WWW console?

Thanks much for your effort and support in Tasmota Project.

arendst, 2017-08-23:
There is a command setoption?? 0/1 to control this. See wiki command list.

Sunrise17, 2017-08-23:
Dear @arendst ,
Thanks for your quick response. SetOption13 in below, solved the issue, it only allows single press with cancelling all any other press modes.

SetOption13 | 1 / on | Enable button single press only for immediate response. Disable by holding 4 x SetOption32

closed 2017-08-23 by Sunrise17