arendst/Tasmota #774: Sonoff RF Bridge 433

Odianosen25, 2017-08-23:
Hello,

Just a question please.

I am trying to build a Mqtt/RF Gateway using NodeMCU. Can I use this Sonoff RF Bridge 433 instead, by just buying it and installing the custom firmware? Or I have to get other kit or software involved.

Many thanks

arendst, 2017-08-23:
It depends what you want to send or receive. It doesn't work with kaku signals

Odianosen25, 2017-08-24:
Many thanks for the response @arendst ,

I run HA, and got this SESOO Wall Pad Glass Panel RF Remote Touch Light Switch (2 way), with an RF remote. 

> http://www.ebay.co.uk/itm/SESOO-Wall-Pad-Glass-Panel-1-2-3-Gang-RF-Remote-Touch-Light-Switch-Black-White-/182489044162?var=&hash=item2a7d31d4c2:m:mfgUKC4dxT5-C6BhPcvqPrQ

I was thinking if it was possible, to just use the new bridge and control them via MQTT>RF that's why I am asking.

Do you think it will be possible?

Kind regards

closed 2018-05-07 by stale[bot]