arendst/Tasmota #775: Sonoff LED Cold and Warm colour switched

connectedthings, 2017-08-23:
Possibly just my device so FYI for others possibly having this issue. Am using 5.6.1, the colour controls are reversed, Cold = Warm and Warm = Cold. Not sure if this is a bug or a HW versioning issue ala Chinese Manufacturing ... 

Checked the wiring on my unit and the pins marked Y for yellow aka warm and W for white aka Cold are correct.

Addressed the issue by switching GPIO_PWM1 and GPIO_PWM2 in sonoff_template.h device definition.

arendst, 2017-08-23:
Did it switch after a firmware upgrade or is this a new device? I did fool around a bit there...

badgerhome, 2017-08-23:
@arendst are you using AiLight library for bulb?

arendst, 2017-08-23:
@badgerhome nope. Sonoff Led is a ser of 4 two colors white led strips.

I'm still waiting for my ailight bulb to remove experimental support.

connectedthings, 2017-08-24:
Not a new unit, did perform an upgrade but can't recall from which version, apologies. It was an upgrade from a 5.x version though

connectedthings, 2017-08-24:
Oh, the version I upgraded from did not have the Cold - Warm slider ....

arendst, 2017-08-24:
So just to make it clear for me:

1 - When you use the slider and move it to the left (Cold) you do receive Warm light?
2 - When you use command Color 2000 you do receive Warm light?

The fact is I did change color control starting with version 5.5.2c. But in my situation it still works as expected:

1 - Slider left is Cold, right is Warm
2 - Color 2000 is Cold, Color 0020 is Warm as it was in version 5.2.0

Are there more people who experience a color swap from Cold to Warm since version 5.5.2c?

connectedthings, 2017-08-24:
Theo, correct moving left (Cold) I get more warm light and more cold when I slide right (Warm)

I have not tried it via commands yet, will try this eve when back at device

connectedthings, 2017-08-25:
Just managed to check this via commands, its also reversed 0020 is Cold, 2000 is Warm

arendst, 2017-08-25:
Do you know if the previous version did it that way too or the reverse? 

In my case both the previous and the latest release work as (I) expected; 0020 is Warm and 2000 is Cold.

connectedthings, 2017-08-25:
The previous version did not have colour control, only brightness. I am starting to think this is an upgrade issue? What is the best way to reset everything? Flash blank.bin and reflash 5.6.x?

arendst, 2017-08-25:
No. The easiest way is to open the webpage and upload 5.6.x via the upgrade webpage. No need to flash a blank.bin

connectedthings, 2017-08-25:
Have done that twice now, reverted my sonoff template change to fix the issue so that I could test the commands as requested. Have just flashed again with the fix.
Could it be changes settings structure between the versions that did not have colour control and those that did? Especially that I am not sure what version was upgraded from to get to 5.6.1

arendst, 2017-08-25:
Color control was always there for the Sonoff Led. Using color CCWW you were able to control both leds in addition to dimmer 1-100. 

The only difference for the Sonoff Led is that it now has color control on the webpage too and that reading the color values has changed but the order should have stayed the same.

arendst, 2017-08-25:
Try to upload sonoff.bin from release 5.5.2 and see what happens. Your settings should stay the same.

connectedthings, 2017-08-25:
Still reversed in 5.5.2, am going to upload a sketch to check GPIO12 & 14 match the output markings on the physical device!

connectedthings, 2017-08-26:
As always the simplest answer is usually the right one, the unit was obviously not tested in the factory, although connections on the device are correct, the warm and cold wires are reversed in the plug connecting to the led strips. 

Apologies for taking bandwidth on this, hopes it helps others should they find themselves here ..

closed 2017-08-26 by connectedthings