arendst/Tasmota #776: BUG - Pulsetime and SetOption14 values lost on reboot

giig982, 2017-08-23:
I'm trying interlock mode (setOption14) with v5.6.1 on Sonoff Dual in order to use Sonoff Dual for rolling shutters as stated in #721 and I can set Option14 and Pulsetime values via web interface, but on Sonoff reboot those values are lost, even if I send a `saveData` command, or `set SaveData 1` for autosave changes.

How to reproduce:
- send `SetOption14  On`
- send `Pulsetime<1> n`
- send `Pulsetime<2> m`
- send `SaveData 1`

Let's try that everything works as expected:
- switch relais <1> ON and let it OFF automatically after n seconds, as set by Pulsetime 
- switch relais <2> ON and let it OFF automatically after m seconds, as set by Pulsetime
- if you switch relais <1> ON while relais <2> is ON, relais <2> turns OFF and relais <1> turns ON and it will turn OFF automatically after n seconds

Then:
- reboot Sonoff
- switch <x> relais ON and it never shut down automatically 
- switch <y> relais ON and it never shut down automatically 

-> Pulsetime and SetOption14 settings are lost.

arendst, 2017-08-23:
Pls provide output of command status 0

giig982, 2017-08-23:
This is the output of `Status 0` after reboot:

`STATUS = {"Status":{"Module":5, "FriendlyName":"sonoff_tapp1", "Topic":"sonoff_tapp1", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":19200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":1, "Sleep":0, "BootCount":17, "SaveCount":95, "SaveAddress":"FB000"}}
STATUS2 = {"StatusFWR":{"Program":"5.6.1", "BuildDateTime":"2017-08-18T16:09:25", "Boot":4, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
STATUS3 = {"StatusLOG":{"Seriallog":0, "Weblog":2, "Syslog":0, "LogHost":"192.168.5.250", "SSId1":"martinRoutero", "SSId2":"milano20125", "TelePeriod":300, "Option":"55800009"}}
STATUS4 = {"StatusMEM":{"ProgramSize":421, "Free":580, "Heap":29, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}}
STATUS5 = {"StatusNET":{"Hostname":"sonoff_tapp1-0826", "IPaddress":"192.168.5.134", "Gateway":"192.168.5.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.5.1", "Mac":"60:01:94:1E:23:3A", "Webserver":2, "WifiConfig":3}}
STATUS6 = {"StatusMQT":{"Host":"192.168.5.248", "Port":1883, "ClientMask":"sonoff_tapp1", "Client":"sonoff_tapp1", "User":" ", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
STATUS7 = {"StatusTIM":{"UTC":"Wed Aug 23 21:05:11 2017", "Local":"Wed Aug 23 22:05:11 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
STATUS10 = {"StatusSNS":{"Time":"2017-08-23T22:05:11"}}
STATUS11 = {"StatusSTS":{"Time":"2017-08-23T22:05:11", "Uptime":1, "Vcc":3.196, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"martinRoutero", "RSSI":84, "APMac":"C0:4A:00:49:E8:B2"}}}`

And this after sending `SetOption14 on` and `Pulsetime<1> 50`:

`STATUS = {"Status":{"Module":5, "FriendlyName":"sonoff_tapp1", "Topic":"sonoff_tapp1", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":19200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":1, "Sleep":0, "BootCount":17, "SaveCount":95, "SaveAddress":"FB000"}}
STATUS2 = {"StatusFWR":{"Program":"5.6.1", "BuildDateTime":"2017-08-18T16:09:25", "Boot":4, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
STATUS3 = {"StatusLOG":{"Seriallog":0, "Weblog":2, "Syslog":0, "LogHost":"192.168.5.250", "SSId1":"martinRoutero", "SSId2":"milano20125", "TelePeriod":300, "Option":"55804009"}}
STATUS4 = {"StatusMEM":{"ProgramSize":421, "Free":580, "Heap":27, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}}
STATUS5 = {"StatusNET":{"Hostname":"sonoff_tapp1-0826", "IPaddress":"192.168.5.134", "Gateway":"192.168.5.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.5.1", "Mac":"60:01:94:1E:23:3A", "Webserver":2, "WifiConfig":3}}
STATUS6 = {"StatusMQT":{"Host":"192.168.5.248", "Port":1883, "ClientMask":"sonoff_tapp1", "Client":"sonoff_tapp1", "User":" ", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
STATUS7 = {"StatusTIM":{"UTC":"Wed Aug 23 21:13:51 2017", "Local":"Wed Aug 23 22:13:51 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
STATUS10 = {"StatusSNS":{"Time":"2017-08-23T22:13:51"}}
STATUS11 = {"StatusSTS":{"Time":"2017-08-23T22:13:51", "Uptime":1, "Vcc":3.196, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"martinRoutero", "RSSI":84, "APMac":"C0:4A:00:49:E8:B2"}}}`

Then, if i restart (even via web interface):

`STATUS = {"Status":{"Module":5, "FriendlyName":"sonoff_tapp1", "Topic":"sonoff_tapp1", "ButtonTopic":"0", "Power":3, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":19200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":0, "Sleep":0, "BootCount":17, "SaveCount":95, "SaveAddress":"FB000"}}
STATUS2 = {"StatusFWR":{"Program":"5.6.1", "BuildDateTime":"2017-08-18T16:09:25", "Boot":4, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
STATUS3 = {"StatusLOG":{"Seriallog":0, "Weblog":2, "Syslog":0, "LogHost":"192.168.5.250", "SSId1":"martinRoutero", "SSId2":"milano20125", "TelePeriod":300, "Option":"55800009"}}
STATUS4 = {"StatusMEM":{"ProgramSize":421, "Free":580, "Heap":29, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}}
STATUS5 = {"StatusNET":{"Hostname":"sonoff_tapp1-0826", "IPaddress":"192.168.5.134", "Gateway":"192.168.5.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.5.1", "Mac":"60:01:94:1E:23:3A", "Webserver":2, "WifiConfig":3}}
STATUS6 = {"StatusMQT":{"Host":"192.168.5.248", "Port":1883, "ClientMask":"sonoff_tapp1", "Client":"sonoff_tapp1", "User":" ", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
STATUS7 = {"StatusTIM":{"UTC":"Wed Aug 23 21:15:37 2017", "Local":"Wed Aug 23 22:15:37 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
STATUS10 = {"StatusSNS":{"Time":"2017-08-23T22:15:37"}}
STATUS11 = {"StatusSTS":{"Time":"2017-08-23T22:15:37", "Uptime":0, "Vcc":3.197, "POWER1":"ON", "POWER2":"ON", "Wifi":{"AP":1, "SSID":"martinRoutero", "RSSI":84, "APMac":"C0:4A:00:49:E8:B2"}}}`

arendst, 2017-08-23:
You are most probably running the sonoff-minimal version. This version, as documented in the release description and wiki is supposed to be used at intermediate loader. It does not save any setting by desgn.

Get yourself the full sonoff version as released and all  your issues are gone.

Have a nice day.

giig982, 2017-08-23:
**Shame on me!**

I uploaded minimal version in order to upload then the normal version, but in the rush of try interlock i forgot to do the second upload.

Sorry for the false issue.

closed 2017-08-23 by giig982