arendst/Tasmota #779: auto off timer

SHatzl, 2017-08-24:
I like tasmota, but I would really need a auto-off timer function for the Sonoff TH. For security reasons I need a function, that switch the relays after X seconds on state back to off.
Means, if i switch it on, it should automatically switch off after X seconds. But the timer should restart if I send a on command within the X seconds, so that the relays can keep alive, if sending on commands.

For me it would be enough to have a variable in user_config for setting the X seconds, i don't need to configure it over the web-interface.

Can anybody help me to include it?

arendst, 2017-08-24:
Well what about command pulsetime?

See wiki commands how to enter the correct time. While active a new power 1 will restart the pulsetime.

SHatzl, 2017-08-24:
pulstime shows no effect for me.. I entered 110 as value for testing. That should instantly switch back to off, right?
But nothing happens.. ):

arendst, 2017-08-24:
After setting pulsetime you'll have to power off/on once to get it starting.

SHatzl, 2017-08-24:
I turned on/off. still no effect. Should i try to reset configuration over web interface?

arendst, 2017-08-24:
No need to reset config.

What is the result of command status 0?

SHatzl, 2017-08-24:
Here is the result.. I'm a little bit confused, because PowerRetain should also be 1, in user_config it's set to 1..

00:04:45 MQTT: stat/heatingwohnzimmer/STATUS = {"Status":{"Module":4, "FriendlyName":"Sonoff", "Topic":"heatingwohnzimmer", "ButtonTopic":"0", "Power":1, "PowerOnState":0, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
00:04:45 MQTT: stat/heatingwohnzimmer/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":1, "Sleep":0, "BootCount":21, "SaveCount":77, "SaveAddress":"F7000"}}
00:04:45 MQTT: stat/heatingwohnzimmer/STATUS2 = {"StatusFWR":{"Program":"5.6.1", "BuildDateTime":"2017-08-24T15:08:03", "Boot":4, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
00:04:45 MQTT: stat/heatingwohnzimmer/STATUS3 = {"StatusLOG":{"Seriallog":2, "Weblog":2, "Syslog":0, "LogHost":"domus1", "SSId1":"WIFI_HTCTRL", "SSId2":"", "TelePeriod":300, "Option":"55800009"}}
00:04:45 MQTT: stat/heatingwohnzimmer/STATUS4 = {"StatusMEM":{"ProgramSize":477, "Free":524, "Heap":27, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}}
00:04:45 MQTT: stat/heatingwohnzimmer/STATUS5 = {"StatusNET":{"Hostname":"heatingwohnzimmer-3988", "IPaddress":"192.168.99.20", "Gateway":"192.168.99.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.99.1", "Mac":"2C:3A:E8:07:2F:94", "Webserver":2, "WifiConfig":2}}
00:04:45 MQTT: stat/heatingwohnzimmer/STATUS6 = {"StatusMQT":{"Host":"192.168.99.1", "Port":1883, "ClientMask":"DVES_%06X", "Client":"DVES_072F94", "User":"homeassistant", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
00:04:45 MQTT: stat/heatingwohnzimmer/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:04:45 1970", "Local":"Thu Jan 01 00:04:45 1970", "StartDST":"Thu Jan 01 00:00:00 1970", "EndDST":"Thu Jan 01 00:00:00 1970", "Timezone":1}}
00:04:45 MQTT: stat/heatingwohnzimmer/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:04:45", "AM2301":{"Temperature":21.8, "Humidity":59.7}, "TempUnit":"C"}}
00:04:45 MQTT: stat/heatingwohnzimmer/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:04:45", "Uptime":1, "Vcc":3.171, "POWER":"ON", "Wifi":{"AP":1, "SSID":"WIFI_HTCTRL", "RSSI":62, "APMac":"00:0F:60:09:EF:CA"}}}

arendst, 2017-08-24:
Why do you edit user_config.h to get these settings set? No need for it. Just use commands and it will be saved in flash.

I see no strange things in your status messages but reading text like "homeassitant" normally means a lot of uncontrolled MQTT messages to be handled....

How did you test the pulstime functionality? Using the TH webpage or Homeassistant?

Just to let you know, my TH with version 5.6.1 and pulse time set to 110 powers off the relay after 11 seconds as expected. Pressing the on button extends the pulsetime as expected too.

SHatzl, 2017-08-24:
You are right, setting it over /cmnd/heatingwohnzimmer/PulseTime1 110 worked like a charm!

I tried both, via homeassistant and via web interface.

Thank you for your help!

closed 2017-08-24 by arendst