arendst/Tasmota #780: Sonoff SC Individual Data commands

EmilySmith0, 2017-08-24:
Sonoff SC publishes MQTT data as
**MQTT**: tele/sonoffSC/SENSOR = {"Time":"2017-08-24T5:42:45", "Temperature":24.0, "Humidity":53.0, "Light":100, "Noise":10, "AirQuality":100, "TempUnit":"C"}
Is it possible to get individual values after modifying code
i,e stat/sonoffSC/state 20 for temperature
stat/sonoffSC/state 21 for humidity
stat/sonoffSC/state 22 for Light
If yes
Where I should I start. I am not a pro :)

arendst, 2017-08-24:
Examine file xdrv_snfsc.ino

EmilySmith0, 2017-08-24:
I know these variables contains values from sensors
[image]

but where I should modify to get get values with commands
like stat/sonoffSC/Status  20
Many Thanks
Emily

arendst, 2017-08-24:
File sonoff.ino around line 1850

EmilySmith0, 2017-08-25:
Sonoff SC publishes MQTT data as
MQTT: tele/sonoffSC/SENSOR = {"Time":"2017-08-24T5:42:45", "Temperature":24.0, "Humidity":53.0, "Light":100, "Noise":10, "AirQuality":100, "TempUnit":"C"}
I want to Publish data as 
tele/sonoffSC/**SENSOR1** = {Temperature":24.0}
tele/sonoffSC/**SENSOR1** = {Temperature":24.0}
tele/sonoffSC/**SENSOR1** = {Temperature":24.0}
tele/sonoffSC/**SENSOR1** = {Temperature":24.0}

EmilySmith0, 2017-08-31:
Changed highlighted line and applied loop for 5 times to get individual values

closed 2017-08-31 by EmilySmith0