arendst/Tasmota #781: Is there an internal variable I can use to see uptime?

stefanbode, 2017-08-24:
Many of my devices run on battery, I want to see how long they are online until they go back into deepsleep. Is there a variable I can use for this?

davidelang, 2017-08-24:
I don't think tasmota does deepsleep, I think it's always running to stay 
connected to the network and look for commands/buttonpresses.

stefanbode, 2017-08-25:
No worries about that. This is my own mod/fork. Working well because I have several solar power WEMOS irrigation sensors. Anyhow is there a variable for the uptime?

arendst, 2017-08-25:
sonoff.ino line 2047

stefanbode, 2017-08-25:
I had to do some changes and use the rtcMem. This ensure I get the runtime, even, if the device reboots or has a blocking loop. Now everything is fine and I always get the total runtime, even if there is a reboot in between. I do see that on the mqttClient.loop(); sometimes it takes up to 5 seconds until the procedure comes back. Sometimes directly. Even on the reboot procedure, there is no reproducible bahaviour. Not critical, but strange. If you enable the DEBUG_THEO you can see that the loop "hang".

closed 2017-08-26 by stefanbode