arendst/Tasmota #785: WS2812 dimmer issue on power

ardeus-ua, 2017-08-25:
I have Sonoff Basic connected to 5V 10A power source and WS2812 connected to GPIO14.
Once I toggle the power state through MQTT, led strip doesn't lights up. \
Checking the console gave me the understanding that LED strip won't light up until I change the Dimmer value (sending the same dimmer value as before the power on). 
Could you please help me if there is a way to retain dimmer value.

TASMOTA v.5.6.1c, HomeAssistant 0.51.2

ionciubotaru, 2017-08-27:
Try `backLog` to send power on and dimmer value on the same mqtt message

ardeus-ua, 2017-08-27:
It doesn't changes if I send the current value to Sonoff via MQTT. I need to send different value in order for sonoff to light up

ionciubotaru, 2017-08-27:
Send to sonoff any value you like, not the current value

ardeus-ua, 2017-08-28:
That's not a best solution. I cannot send the current value to keep it on the same comfortable level or I have to use random number each time.

closed 2018-05-07 by stale[bot]