arendst/Tasmota #787: APP_TIMEZONE 99 does not work

ramoun, 2017-08-25:
I've compiled 5.3.0 using 
#define TIME_DST               Last, Sun, Oct, 1, +660  // Last sunday in march at 02:00 +120 minutes
#define TIME_STD               Last, Sun, Apr, 1, +600   // Last sunday in october 02:00 +60 minutes
// -- Application ---------------------------------
#define APP_TIMEZONE           99                // [Timezone] +1 hour (Amsterdam) (-12 .. 12 = hours from UTC, 99 = use TIME_DST/TIME_STD)

But when flashed it does not appear to make it to the device, I can change it using MQTT but I have a bunch of devices to flash.

If I change 99 to 10 (AEST) compile flash I see the same behaviour, is this a known bug or just my install

arendst, 2017-08-26:
Hint: Read the header of the file.

ramoun, 2017-08-26:
Thanks Theo, this explains the behaviour and few other things ;-)

All credit to you for taking the time to reverse engineer, build and maintain the code, as well as taking the time to answer the obvious questions.

closed 2017-08-26 by ramoun