arendst/Tasmota #790: Please Update local PubSubClient to get rid of a timeout

stefanbode, 2017-08-26:
Hi Arend,

there is a change discussed (should be committed) and works fine for me in the PubSubClient.cpp Line 209ff

I recommend doing the change because this avoids some "Blocked loop" and speed up start time significant.

https://github.com/knolleary/pubsubclient/pull/314

```
// reads a byte into result
boolean PubSubClient::readByte(uint8_t * result) {
   uint32_t previousMillis = millis();
   if(!_client->available()) {
       return false;
   }
   *result = _client->read();
   return true;
}
```

stefanbode, 2017-09-25:
Instead of the yield(); we can also use a delay(1); It has the same effect. The whole change is to avoid a blocking of the read procedure until the timeout(). With the delay(1) the ESP has the possibility to get more bytes from the interface and end successfully. Without it, it will never get the initially missing byte from the channel.

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst