arendst/Tasmota #791: Flash count goes up every time I switch

revdaz, 2017-08-26:
Hello,

I just see that in the debug view (because some problems with wall switches) of the sonoff website that everytime the flash counter goes up when I switch with a switch on GPIO14. Is this the right behavior?

console:
20:07:11 MQTT: stat/sonoff_rf_esszimmer/POWER = OFF
20:07:12 Cnfg: Save (1648 bytes) to flash at F5 and count 215

Is there no problem with wearing of the flash?

arendst, 2017-08-26:
If you do not care if the relay will be in the same state after power on of sonoff you can play with commands setoption0 and savedata.

revdaz, 2017-08-26:
Thank you for the quick answer!
if it's no problem for the sonoff flash, for me it's ok :-)

You really do a great work with this project!

closed 2018-05-07 by stale[bot]