arendst/Tasmota #792: Flashing without soldering and using putty (Knete in german)

revdaz, 2017-08-26:
Hello,

this is no problem, just a tip, but I don't know where to publish....

I figured out, that it's very easy to connect the flasher cables without soldering and using car cleaning putty (Lackreinigerknete in german), this is very strong but I think it should also work with normal putty.
Just fix the connected cables with pin header temporary with the board, after flashing take it away again.
 
I flashed with this method already some sonoff S20, RF and also even sonoff Touch without soldering any cable. Only if you will need the GPIO's you have to solder.

Maybe somebody can put this topic in the right folder or integrate it in the wiki when thinking this will be useful.
[image]
[image]

x3r4n0, 2017-08-27:
Kannst du mir vllt verraten wie die PIN Belegung bei dem Sonoff T1 aussieht?

revdaz, 2017-08-27:
Dei Belegung des 4 PIN Headers ist gleich wie bei allen sonoffs, das orangene Kabel ist bei mir Ground, leider habe ich sonst keine Bilder gemacht und der Schalter ist eingebaut. aber auf: https://forum.fhem.de/index.php?topic=64804.0 sieht man es eigentlich auch ganz gut, für die Bilder dort zu sehen musst du dich aber anmelden. um in den Flashmodus zu kommen musst du noch GPIO brücken bevor du ihn mit strom versorgst, dazu hab ich das einzelne gelbe kabel genommen, dass bei mir auf dem breadbord auf GND liegt und an das ESP Modul gehalten, zweiter Kontakt unten rechts.

pascalsaul, 2017-08-28:
You can do it even more easier and quicker by just holding the pin header in your hand during flashing :) No clay needed ;)

titomontes, 2017-08-29:
And you could also publish in english; it´s really rude to to publish in other languages.

longbikez, 2017-09-03:
Where do I find this header with a 1.5mm pitch, I've looked everywhere I can think of.

revdaz, 2017-09-04:
Hello,

I think there is a misunderstanding, you will need 2,54 mm headers.... (maybe its 1,5 in inch?) you can order them by ebay or directly by Amazon. e.g. https://www.amazon.de/gp/product/B01LWD5MPB/ref=oh_aui_detailpage_o08_s00?ie=UTF8&psc=1 (but they have longer delivery time.

longbikez, 2017-09-05:
Yup, I don't know what I was thinking.

jetema, 2017-09-08:
I don't use soldering, I just use a 4 pin male connector , it works good 
[image]

longbikez, 2017-09-08:
Thanks for the pic, it helps. Do you use Blynk boilerplate?

jetema, 2017-09-09:
No, I'm sorry, I use all my sonoffs through web browser mainly

closed 2018-05-07 by stale[bot]