arendst/Tasmota #795: Sonoff SC not accurate

ajcorrea, 2017-08-27:
Hello,

I´m testing Tasmota with Sonoff SC, but the returned info are very inacurate !!

Using a Thermo-Hygrometer (very accurate)
Temp: 24.2 C
Hum: 37%

On the sonoff SC:

Sonoff SC Module

Temperature	29.0°C
Humidity	12.0%

ionciubotaru, 2017-08-27:
Sonoff SC has 2 parts:
First based on ATMEGA328 which collects data from sensors and send it to ESP8266 via serial interface
Second based on ESP8266 which get data from serial and send it to MQTT server via WIfi

Tasmota is responsible only for est8266 part, so It has no relation to sensor reading and calibration.
The software for atmega328 can be found on itead website, try to update your SC to the latest version and, if errors persist feel free to send them any bugs you found

arendst, 2017-08-27:
#454

helmar74, 2018-03-13:
Same issue here,  Humidity only meassured between 10-15% (which is completely wrong), Temperature about 5 degrees to high.
Has anyone found a solution to get accurate values from atmel?
e.g new firmware for atmega328P?

Tjhanks a lot

bartaspoz, 2018-03-18:
Is replacing the DTH11 with DTH22 would do the job? They are electrically identical, so the replacement should be straight forward

ionciubotaru, 2018-03-18:
Warning ! DHT11 Power supply: DC 3.5～5.5V - https://akizukidenshi.com/download/ds/aosong/DHT11.pdf
3V3 is not suitable.

bartaspoz, 2018-03-19:
But this guy was able to switch DTH11 with DTH22:
https://www.youtube.com/watch?v=EO53fWGTFUo

ionciubotaru, 2018-03-19:
It's possible, I'm not sure how accurate is the reading.

bartaspoz, 2018-03-19:
I have the DTH22 connected to Sonoff Basic. It is quite accurate, when compared to a weather station from a local shop readings are almost the same. And the resolution of temperature and humidity is 0.1% compared to 1% in DTH11

Joshfindit, 2018-03-31:
This seems like a good thread to talk about making the SC more accurate.
As far as the video, it looks like going to the DHT22 is straightforward, and https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration shows that it is supported by Tasmota.

Is there anything that can be done about the air quality sensor?
Better resolution, track CO2 separate from dust, check oxygen levels, etc?

samlowry, 2018-05-23:
@arendst it's not related to DHT11 resolution anyhow. Look at the pic, I bought a several of devices and stoped my smarthome project because after flashing them with your firmware they shows a very irrelevant information, look at the pic:
[image]

It's unusable. At all. Everything is wrong. And DHT11 isn't the cause - it has 5% accuracy (for 28° it means ±0.05°) and it's work proper with default firmware (I compare to gm320 infrared thermometer - default firmware pretty close to thermometer, ±2°) when your firmware shows +5° all the time. I didn't research is it constant delta or not, but looks like in average it +5° (compared to AC settings). As for humidity it's opposite now - default firmware shows 56%, yours - 45%. I opened and experimented with second device only today, before I compared temp with AC and ir thermometer, so not sure about delta of humidity, but pretty sure in temp. - it's always higher on about 4-6°.

So there is a bug maybe?

tlc76, 2018-05-26:
I confirm the same issue here:
- sonoff sc with tasmota shows 35° temperature
- weather station in the same room shows 25°.

Is it possible that in the original ESP firmware maybe some normalization was applied on the AT328 sensor readings before sending out the info via wifi?

arendst, 2018-05-26:
Mine is 3 degrees too hot. As the sensor is in the casing you might expect problems with this hardware setup. 

BTW my humidity is way off too. As iTead also writes the device is a proof of concept and do not expect any thing useful from it:

> Before you place the order, please note:
> - This is more a TOY than a final product. Please take it as a product for hobbyists. 
> - Make sure you know the accuracy of the sensors, especially for the DHT11, microphone, there are some errors.
> - The firmware is not well polished. There may be some bugs.

tlc76, 2018-05-26:
Right, but come on, the device cannot be used not even as a toy. I would have expected some minimal degree of maturity with this product even in the harsh conditions of the disclaimer you've quoted earlier :)

Anyway it seems the easiest thing we could do is upgrade to DHT22, but I'm curious what's the new deviation.

Or, a function could be built and applied to the raw sensor values, which should adjust the reported values based on various sensor readings versus corresponding real values (measured with a "competent" device), for both temperature and humidity. Then the function could be applied directly in the Tasmota firmware, in function SonoffScSerialInput (values   sc_value[0] DHT11 Humidity and sc_value[1] DHT11 Temperature in Celsius).

That's just an idea :)

ascillato, 2018-06-13:
Hi,

If your issue was solved, please close it. Thanks :+1:

helmar74, 2018-06-13:
It's not solved, but can be closed anyway

ascillato2, 2018-06-27:
A calibration now can be done with new commands ADD, SUB and MULT for rules.

Thanks.

closed 2018-06-27 by ascillato2

tlc76, 2018-07-23:
All,

Please see below the Temp and Humidity values as they are being reported by two sets of sensors:
1. green - Xiaomi Aqara Temp/Humidity sensor (high precision)
2. yellow - Sonoff SC device (unreliable).

Both sensors are placed one next to another.

Considering the Xiaomi device has a very good accuracy (I have "visually" tested it against several weather stations), the average offsets of the reported values are:
a. +9 degrees temperature deviation reported by Sonoff SC
b. +19% humidity deviation reported by Sonoff SC

As the acquired sensor value range is quite narrow at this moment, I will continue to monitor the data sets in other conditions as well - most probably I need to wait for the autumn/winter to come :)

Ideally the tests should be performed in lab conditions, to enable measuring of the full range of reported values and hence to allow one to adjust the values reported by Sonoff SC. I will also send an email to our friends from Itead, maybe they will consider performing these tests.

If the deviation problem could be solved, I believe Sonoff SC would prove itself to be a pretty valuable smart home thing, even thought the DHT11 sensor has a limited precision.

[image]
[image]

vontrapp, 2018-09-23:
How is this closed? I didn't see a resolution...

Is an offset calibration really sufficient to get some decent accuracy?

ascillato, 2018-09-23:
@vontrapp

This issue is closed because it is solved. If you want to modify the sensors values inside Tasmota just use the example provided in rules at the wiki. 

Thanks

ilvenos, 2018-10-16:
@ascillato  could you give example comand?
"Sub1 5 " doesnt work for me

ascillato, 2019-01-10:
> what is the name of ther Sensor (Sensorname#temperature).

In order to be used as a trigger in a rule every teleperiod time check the following example:

```
rule1 1
rule1 on tele-SonoffSC#Temperature>25 do power1 on endon
```

matrix77, 2019-10-06:
Hi, anybody can help with the default orig.  itead firmware for sonoff sc,only the esp firmware,thank you.

meingraham, 2019-10-06:
If you plan is to use a binary that someone else provides from one of their devices to flash it back onto your device, the likelihood is extremely high that this will not work. Itead binaries are usually "keyed" to the specific device (i.e., the binary only works with the device it came on).

matrix77, 2019-10-06:
I understand, one try is more than anything.thk.

meingraham, 2019-10-06:
OK then, here you go - https://github.com/rt400/Sonoff-Firmware/blob/master/SONOFF_SC_2.0V.bin

matrix77, 2019-10-06:
Many thanks,i try it but the flashing stop's at 27%, i think is something with the backup,do u have any pot to try it?or test it?

meingraham, 2019-10-06:
This is not a Tasmota issue. That repository link I provided is just that, a repository where users have "donated" their binaries. THat's about it. You and Google are on your own as far as how to attempt to restore the firmware.

How are you trying to load that firmware? If you are trying to load using Tasmota OTA, that's not going to work. You need to use serial flashing, erase_flash, and then upload the file. Also, I have NO idea what settings you should use (e.g., flash size, DIO/DOUT/..., etc.).

Mike

matrix77, 2019-10-07:
Understand,thk you for your help.

matrix77, 2019-10-10:
Hi,i follow your steps i succesuly uploaded an orig. backup firmware from sonoff rf bridge on sc the led is blinking it shows wifi connection that sims Like the uploading method is good, when i upload your sc firmware doesn't do anything.

meingraham, 2019-10-10:
That isn't my firmware file. I don't own a Sonoff SC. It's just a file that 's someone uploaded to that repository.

As your issue is not related to Tasmota, it's best to not spam this issue with any further dialog on your efforts to restore Itead firmware on a Sonoff device. Please take any further questions to the Itead support forum - https://support.itead.cc/

Best of luck with your efforts.

Mike