arendst/Tasmota #798: Sonoff RF double press on button does not go to learn mode

gheesung, 2017-08-27:
I am using tasmota v5.6.1. The sonoff rf does not go into learn mode when double pressed on the switch.

gheesung, 2017-08-27:
I have tried multiple consecutive presses, the LED will turn red but not flashing.

Did I setup the firmware wrongly?

arendst, 2017-08-28:
RF is NOT controlled by Tasmota. If the LED turns red that means that the RF microcontroller is waiting for your RF command. If it cannot recognize your RF remote control it won't flash.

gheesung, 2017-08-28:
I have discovered the problem. I am trying to power the board using the FTDI adapter and it does not have enough current to power both ESP and RF board. 

In the wiki page, the instruction to prepare the Sonoff RF is to solder a jumper across R21. This is not necessary as I also found out that I can just short the inner pin of R21 when powering up the device. It will go into programming mode.

[image]

[image]

closed 2017-08-30 by gheesung