arendst/Tasmota #799: GPIO14 as normal wall switch

titomontes, 2017-08-27:
Hello

I just read severao posts in the forum refering to GPIO14; I just want to use it as a wall switch (as I understand, I have to hook it up to GPIO14 and GND), but having selecting <09 Switch1> in config and switchmode <4> in console, GPIO14 still functions as toggle (shorting and holding returns an ON state, but disengaging does nothing).
I,ve already selected 

> #define SWITCH_MODE            PUSHBUTTONHOLD_INV

before flashing, I don´t know what else to do.

Please help!

ionciubotaru, 2017-08-27:
set `switchTopic` if you don't want to toggle the relay

titomontes, 2017-08-28:
Thanks

I´m referring to using sonoff as standalone: I need to use a physical normal wall switch; Do I need to use 

> SwitchTopic

 in console?, if so, what value do I need to enter?

ionciubotaru, 2017-08-28:
If you need to toggle RELAY1 with SWITCH1 set `switchTopic 0`
If you need to send a MQTT message and not toggle RELAY1 with SWITCH1 set `switchTopic <yourDesiredTopic>` 

Hardware solution: connect your wall switch to GPIO0

titomontes, 2017-08-29:
I´ve seen [this video](https://youtu.be/8mz5sCAvDAY?t=22m36s) a million times. In it, you can clearly see that he uses GPIO14 to detect a state change, specifically with Tasmota. I´m not feeling really comfortable soldering and desoldering, much less adding a resistor (not much ability for it, anyway. Only managed to solder through the holes that already come empty), so I´m searching hour after hour for a solution, with not much success...

PeggyFree, 2017-08-29:
Use a TH10, you'll get 3.3v gnd and pulled up gpio14 through the Jack plug.
No soldering needed.

titomontes, 2017-08-30:
Yeah, a little more expensive and a lot less challenging. Basic works just fine for that, and TH10 also needs to be flashed...

titomontes, 2017-08-30:
Sorry for the double comment...

nizamaj, 2017-09-05:
@titomontes : I had the same issue.... finally i figure it out ...Now it working fine for me , Once everything configured and hooked up ...  find out the IP address of the Sonoff (Tasmota) and log in and then --- Configuration --- Configure Module --- Module type select Sonoff Basic  & Bottom GPIO14 Sensor  Change to 09 Switch 1 and save .. if everything fine it should work  .

lukesaber, 2018-01-10:
This looks like a fantastic project and i'm very thankful for everyone contributing.

I have a question that im surprised i cannot find an answer to...

If i have a toggle switch holding the device on via GPIO14, and i connect to it via smartthings, can i turn it off via wireless communication?
What happens with the state of the switch?
do i need to cycle the physical switch off and then back on to turn the light (or appliance) back on, or is my switch inverted?

hoping someone else has asked and this is old news.

Thanks

davidelang, 2018-01-10:
On Tue, 9 Jan 2018, lukesaber wrote:

> If i have a toggle switch holding the device on via GPIO14, and i connect to it via smartthings, can i turn it off via wireless communication?

yes

> What happens with the state of the switch?

nothing changes, there is no ability to physically move the switch

> do i need to cycle the physical switch off and then back on to turn the light (or appliance) back on, or is my switch inverted?

that depends on you you configure the switch, you can make it so that you have 
to turn the switch off and on, or you can have it that any change in the state 
of the switch toggles the light (in which case the switch doesn't really have an 
off and on, just like a 3-way switch doesn't)

nhorvath, 2018-03-27:
> that depends on you you configure the switch, you can make it so that you have to turn the switch off and on, or you can have it that any change in the state of the switch toggles the light (in which case the switch doesn't really have an off and on, just like a 3-way switch doesn't)

Could you elaborate on what I would need to configure to make a normal on/off toggle switch (non momentary) on GPIO14 work like a 3 way switch? In addition to setting GPIO14 to 09 Switch1 what would I need to do? I'm currently using (and would like to still work) MQTT to toggle it on/off. The built in pushbutton is physically inaccessible in my setup.
Ideally, I'd like the toggle switch to also update MQTT state.
I'm using a sonoff basic.

Thanks for any assistance.

nhorvath, 2018-03-27:
I'm going to answer my own question.
In Module Config set GPIO14 to 09 Switch1
Set switch 1 to PUSHBUTTON_TOGGLE mode, which can be done by opening this url in a browser:
`http://[ip address of sonoff]/cm?cmnd=SwitchMode1%207`
if you have admin password set:
`http://[ip address of sonoff]/cm?user=admin&password=[password]&cmnd=SwitchMode1%207`

Toggling the switch sends the appropriate MQTT message and toggles the relay.

kamenyev, 2018-05-08:
Hello,

I just tried the above mentioned way to connect a "regular" wall switch to GPIO14 on a Sonoff Basic. After saving the changes in the Module Config (set GPIO14 to 09 Switch1) the led/relay went off and since then no response from the board.

I tried to re-flash with platformIO, and it seemingly succeeded, but still no sign of life. No WiFi, and no button response.
Does anyone have an ide what could the reason be?

Thanks in advance.

kamenyev, 2018-05-11:
> try flashing an empty image to clear out all the config area, then
> reflashing. when you flash normally it doesn't overwrite the config section
> of the flash.

Thank you!
I' didn't know how to write empty image, but the latest firmware seemed to
work.

Thanks again.

lukesaber, 2018-05-12:
I don't want to get into gritty what-if's before covering the basics...

When you say you connected your wall switch to the gpio, was the other side
of the wall switch connected to mains voltage or the ground potential of
the sonoff?

Brad

nhorvath, 2018-05-12:
Neither of the 2 on my desk have that on it. It appears to be a capacitor... no idea what it is doing there.

lukesaber, 2018-05-12:
> This wall switch was never connected to mains. The connection was made
> betegen GPIO14 and GND. Adj éven t pins on the added header. The Fradinak
> egy I called it wall switch is to make the impression that it is a switch,
> not a push button.
>
> Btw. I found this SMD component soldered accross two (for me unknown) pins
> on the back side of the PCB. Non of my previous boards had it. [Image]

Two suggestions as I can't look at the image on my phone, I'll have to look
at home...

Add a pull up resistor to the gpio pin from 3v, this will give it a default
state when the switch is not grounding the gpio.

Also, put in a small value capacitor across the gnd/gpio line.

I used a bunch of these as three way switches with the switch about 40'
away, kept getting nuisance on/off events or flickering. Capacitor solved
it.

Good luck!

Brad

nhorvath, 2018-05-12:
> issuecomment-387510724
> jEnIlqHInxks5twe31gaJpZM4PDyOq
> bVxZADuaGkj92Ir6rEuks5txmeYgaJpZM4PDyOq

GPIO 14 has an internal pull up. The pull up resistor should not be
necessary. The cap will help with any bounce in the contacts nor noise on
the line though.

closed 2018-07-11 by stale[bot]

devianpctek, 2018-11-21:
> I'm going to answer my own question.
> In Module Config set GPIO14 to 09 Switch1
> Set switch 1 to PUSHBUTTON_TOGGLE mode, which can be done by opening this url in a browser:
> `http://[ip address of [...]

Great, thanks, I was struggling to get my sonoffs working with a regular switch, now is working fine 😁