arendst/Tasmota #800: Sonoff T1 2CH - How to get into flash mode?

m3zm3riz3, 2017-08-27:
I have connect TX, RX, 3v, GND and hold down button labelled S1 when connecting it to usb however the module powers on and then fails to allow upload. Showing the following error on Platform IO,

"Auto-detected: COM3
Uploading .pioenvs\sonoff\firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error -1
 [ERROR] Took 22.07 seconds

 
 [SUMMARY]
Environment sonoff-minimal      [SKIP]
Environment sonoff-ds18x20      [SKIP]
Environment sonoff              [ERROR]
 [ERROR] Took 22.07 seconds"

Any thinking on the reason or am i not putting the module into flash mode correctly.

closed 2017-08-27 by m3zm3riz3

12voltman, 2017-10-03:
The button on the 2ch model is not connected to GPIO0 like it is on the basic sonoff.  On this board, the switch, as well as the two relays, is connected only to a helper 8051 type microcontroller.  To flash firmware to the ESP8266 onboard and control the relays you will need to do some mods to the board including removing the helper chip, removing the leds, and soldering some small wire to some smd resistors.  I'm working on a full writeup of the process, and in my experience it should take about 20mins with the right equipment.

davidelang, 2017-10-03:
is this a new 2ch version? the original one is supported without hardware 
modifications.

I wouldn't call anything that required unsoldering a chip "supported"

jeylites, 2017-10-03:
I’m not sure how many versions are out there. I have got a T1 3 gang which has an Atmel ic on it.  I’m guessing it’s for the RF 433 control. 

What I found out is, with the Atmel on, you can never get the ESP8255 in to flash  mode. Moreover, the tac button is not connected to GPIO 0.   After a couple hours  of  thinkering, I soldered a wire on GPIO 0  and ground, and  removed  the Atmel ic . It flashed right away. 

It’s quite a step but it works. I’m guessing @12voltman is on a similar line as well?

Bockwurst19, 2017-11-11:
Hi @ALL
Also have a Sonoff T1 Touch.
Could show me where 3v3 RX TX GND and GPIO 0 is on the board.

Thanks MFG

jeylites, 2017-11-11:
What version is your T1?

Bockwurst19, 2017-11-11:
This here
https://www.imagebanana.com/s/873/xcjAO47H.html
https://www.imagebanana.com/s/873/Sy5jGQR5.html

jeylites, 2017-11-11:
@Bockwurst19 

That looks like the latest version, you going to have to trace the GPIO 0 pin. or flash using[ SonOTA](https://github.com/arendst/Sonoff-Tasmota/wiki/SonOTA)

webtop, 2017-12-13:
@jeylites Is the Atmel chip you're talking about, the HD6001? 
I have the following board in a 3-way switch by some random company. The board though does state it is an Itead board, and the WiFi module is the PSF-B85.
I'm trying to flash the ESP using an attached grounded button connected to GPIO0 and no luck.

[image]

jeylites, 2017-12-14:
@webtop the HD6001 is a touch controller which should interfere with the uploading process. However, base on the photo, yours doesn’t seems to have an Atmel installed, a  8 legged IC.

Make sure your  pins are right before uploading . Check the reference picture below
[image]

jimmyw, 2018-01-06:
These pictures show a completely different product that i have, i have no module on mine.

I started to reverse engineer my Sonoff T1 switch here:

https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1

I noticed the helper-chip controls both the GPIO0 and the RESET line on the esp, have not got it to flash just yet, but im on it. There must be some way to let the silabs chip get the esp to enter download mode, just need to find it.

webtop, 2018-01-11:
Yeah, I'm the same...this is the one device I haven't yet flashed, although I might just connect direct to the ESP and power it

inter-faced, 2019-03-21:
I flashed the Sonoff T1 EU 2CH by connecting GROUND to R19 resistor (without soldering). Just follow the explanation in the video https://www.youtube.com/watch?v=0G0JA0Bc54k