arendst/Tasmota #801: Improvement `switchMode 5`

ionciubotaru, 2017-08-27:
Is it possible to have `switchMode 5` sending `ON` and `OFF` to `switchTopic` every time the switch status changed ?
Using this information the HA software can decide what to do.
Of course the SENSOR telemetry message do this job, but those messages has at leat 10 s delay.

I like to play the bell1.mp3 when the door is open and the bell2.mp3 when the door is closed

nightcat91, 2017-08-29:
It sounds like you want to get "On" and "Off" as the commands that you receive over mqtt. 
In that case maybe Switch Mode 1 or 2 would better suit your needs. Give that a try
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#main

ionciubotaru, 2017-09-08:
Thank you @nightcat91 `switchMode1` is the solution

closed 2017-09-08 by ionciubotaru