arendst/Tasmota #803: Hue Emulation not detected by Alexa

m3zm3riz3, 2017-08-27:
I have a Sonoff T1 2Ch, which is set to Hue Emulation but Alexa cannot discover the devices; in the log it showa;

21:27:14 Wifi: Checking connection...
21:27:14 Wifi: Connected
21:27:22 UPnP: HUE 3 response packets sent to 192.168.1.87:50000
21:27:22 UPnP: HUE 3 response packets sent to 192.168.1.87:50000
21:27:23 UPnP: HUE 3 response packets sent to 192.168.1.90:50000
21:27:23 UPnP: HUE 3 response packets sent to 192.168.1.90:50000
21:27:34 Wifi: Checking connection...
21:27:34 Wifi: Connected

But is never discovered, however i can switch emulated bridge is discovered by the hue up and so are the lights.  Any hints on whats wrong, has it been an Alexa update thats screwed the devices discovery of emulated hue devices?

Pa-De, 2017-08-28:
Hi m3zm3riz3,

same issue with my Sonoff 4ch. WeMo is recognized by my Echo dot, but not the hue bridge. Then I tried to implement it into openHAB (hue binding). It recognizes the Sonoff but there is an error that user is already defined.

Courty40, 2017-08-29:
#698 has a thread mentioning the same issue, it appears to be an update on Alexa that wont not detect the Sonoff in Hue mode

Courty

closed 2018-05-07 by stale[bot]