arendst/Tasmota #804: Unable to compile without WS2812

yoakim82, 2017-08-27:
When I try to Verify/compile the sonoff.ino sketch in the Arduino IDE (using the recommended non-admin install, dedicated to this project) I get error messages which I think are related to the WS2812 driver. However I did not choose to include WS2812 support (by adding // in front of the three WS218-related defines in user_config.h). Did I miss some crucial library? I believe I've followed the wiki instructions firmly.

Any help is highly appreciated.

Cheers,

Joakim

---------

In function 'void sl_rgb2hsb(float*, float*, float*)':

xdrv_snfled:377: error: 'RgbColor' was not declared in this scope

   RgbColor dcolor;

   ^

xdrv_snfled:377: error: expected ';' before 'dcolor'

   RgbColor dcolor;

            ^

xdrv_snfled:380: error: 'dcolor' was not declared in this scope

   dcolor.R = sl_dcolor[0];

   ^

xdrv_snfled:383: error: 'HsbColor' was not declared in this scope

   HsbColor hsb = HsbColor(dcolor);

   ^

xdrv_snfled:383: error: expected ';' before 'hsb'

   HsbColor hsb = HsbColor(dcolor);

            ^

xdrv_snfled:384: error: 'hsb' was not declared in this scope

   *hue = hsb.H;

          ^

sketchbook\sonoff\xdrv_snfled.ino: In function 'void sl_setHSB(float, float, float, uint16_t)':

xdrv_snfled:425: error: 'HsbColor' was not declared in this scope

   HsbColor hsb;

   ^

xdrv_snfled:425: error: expected ';' before 'hsb'

   HsbColor hsb;

            ^

xdrv_snfled:443: error: 'hsb' was not declared in this scope

       hsb.H = hue;

       ^

xdrv_snfled:446: error: 'RgbColor' was not declared in this scope

       RgbColor tmp = RgbColor(hsb);

       ^

xdrv_snfled:446: error: expected ';' before 'tmp'

       RgbColor tmp = RgbColor(hsb);

                ^

xdrv_snfled:447: error: 'tmp' was not declared in this scope

       sl_dcolor[0] = tmp.R;

                      ^

exit status 1
'RgbColor' was not declared in this scope

arendst, 2017-08-29:
You still need to provide the NeoPixelBus library as documented in the (just updated) wiki.

yoakim82, 2017-09-01:
Thanks,

I did install it as one of my attempts to mitigate the error, however I still got these errors. Then I tried to launch the sonoff arduino sketch/project från the cloned git folder rather than following the "copy the sonoff folder to your sketchfolder".

Maybe it's just me not being used to now things are organized within the arduino realm.

Anyway, thanks for taking the time to help out.

Cheers,

Joakim

arendst, 2017-10-30:
With version v5.9.0 no need to install neopixeldbus library if no ws2812 is used.

closed 2017-10-30 by arendst