arendst/Tasmota #806: Only first topic on subscribe cmd/xxxx/# (MQTT) during startup gets executed

stefanbode, 2017-08-28:
If there is only ONE retained topic, everything is mostly fine. Only if there are like in the example 4 topics, the TASMOTO only works on the first and the other 3 got ignored. Maybe a timing problem. Maybe the QOS 0 is the problem. I played around with the .loop() but with no success. Can someone reproduce?

```
1503911097:     hm/setting/ESP_2DB1E1/# (QoS 0)
1503911097: DVES_2DB1E1 0 hm/setting/ESP_2DB1E1/#
1503911097: Sending SUBACK to DVES_2DB1E1
1503911097: Sending PUBLISH to DVES_2DB1E1 (d0, q0, r1, m0, 'hm/setting/ESP_2DB1E1/deepsleep', ... (1 bytes))
1503911097: Sending PUBLISH to DVES_2DB1E1 (d0, q0, r1, m0, 'hm/setting/ESP_2DB1E1/webserver', ... (1 bytes))
1503911097: Sending PUBLISH to DVES_2DB1E1 (d0, q0, r1, m0, 'hm/setting/ESP_2DB1E1/status', ... (1 bytes))
1503911097: Sending PUBLISH to DVES_2DB1E1 (d0, q0, r1, m0, 'hm/setting/ESP_2DB1E1/power', ... (1 bytes))
1503911097: Received SUBSCRIBE from DVES_2DB1E1
1503911097:     hm/setting/sonoffs/# (QoS 0)
1503911097: DVES_2DB1E1 0 hm/setting/sonoffs/#
1503911097: Sending SUBACK to DVES_2DB1E1
1503911097: Received PUBLISH from DVES_2DB1E1 (d0, q0, r0, m0, 'hm/status/ESP_2DB1E1/RESULT', ... (21 bytes))
1503911097: Sending PUBLISH to stbode_113852758 (d0, q0, r0, m0, 'hm/status/ESP_2DB1E1/RESULT', ... (21 bytes))
1503911097: Received PINGREQ from ESPCLIENT_001C0A
1503911097: Sending PINGRESP to ESPCLIENT_001C0A
1503911098: Received SUBSCRIBE from DVES_2DB1E1
1503911098:     cmnd/DVES_2DB1E1/# (QoS 0)
1503911098: DVES_2DB1E1 0 cmnd/DVES_2DB1E1/#
1503911098: Sending SUBACK to DVES_2DB1E1
```

arendst, 2017-08-28:
Maybe you'll have to wait for the last of three subscribes to finish before sending your publish bombardement; The code is not in a state to service your publishes yet.

stefanbode, 2017-08-28:
I added some code after the first and upcomming subscribes to exactly address this issue, but I only got back the first message. Code looks like this: Normally I expect the callback() to gather these messages because they should be in the "buffer". 

```
    mqtt_publish_topic_P(0, PSTR("POWER"), svalue);
    getTopic_P(stopic, 0, sysCfg.mqtt_topic, PSTR("#"));
    mqttClient.subscribe(stopic , 1);
    for (byte i=0; i<255; i++) {
      mqttClient.loop();  // Solve LmacRxBlk:1 messages
      delay(10);
    }
    if (strstr(sysCfg.mqtt_fulltopic, MQTT_TOKEN_TOPIC) != NULL) {
      getTopic_P(stopic, 0, sysCfg.mqtt_grptopic, PSTR("#"));
      mqttClient.subscribe(stopic, 1);
      for (byte i=0; i<255; i++) {
        mqttClient.loop();  // Solve LmacRxBlk:1 messages
        delay(1);
      }
      fallbacktopic = 1;
      getTopic_P(stopic, 0, MQTTClient, PSTR("#"));
      fallbacktopic = 0;
      mqttClient.subscribe(stopic, 1);
      for (byte i=0; i<255; i++) {
        mqttClient.loop();  // Solve LmacRxBlk:1 messages
        yield();
      }
    }
```

arendst, 2017-08-28:
Tasmota does not queue MQTT subscribes.

stefanbode, 2017-08-28:
I have ready the 40ms issue, that ESP is not suitable to read messages coming in that fast. Lets check if we can get the subscribe move to QoS1. This should help

stefanbode, 2017-08-28:
Ok, I replaced the setCallback with a smaller one to see, if there is a timing problem. But there is no one. Only the first message arrived. Therefore the problem is deeper in the SDK. Mosquito allows some config on the subscription to use QoS, but this does not work for the initial set. Therefore also no help here. I would suggest a note, that the subscriber will not work on retained messages. And if, only of the first send by the MQTT broker.

closed 2017-08-28 by stefanbode