arendst/Tasmota #807: Is it possible to reduce the antenna output power?

fttx, 2017-08-28:
Or set the wifi mode to something lower than 802.11n?

stefanbode, 2017-08-28:
The ESP8266 support IEEE 802.11 b/g/n Wi-Fi. Normally the client should automatically choose the best he can do with the access point. I don't know an option where you can influence it when running the ESP8266 in access point mode. Lowering the antenna output, e.g. save energy is not possible. afaik.

davidelang, 2017-08-28:
switching out of n mode will only make things worse, never better

fttx, 2017-08-28:
I'm asking this because i'm getting some false positive with a PIR sensor, and i read in a forum that lowering the wifi power improves the situation.  
I think that the mode can be changed through WiFi.setPhyMode() function, but, is there any option in the user_config_override that allows to do this?

davidelang, 2017-08-28:
If you have interference from wifi or other sources, add a small capacitor 
between the PIR and ground.

fttx, 2017-08-28:
Between the PIR out and ground?
I already have a big capacitor between VCC and ground, that is shared from both ESP8266 and PIR.
And i also have a pull down resistor on the PIR out, and a 220nF capacitor across pin12 und pin13 of the BISS0001. And i'm still getting false positives.

davidelang, 2017-08-28:
yes, between the GPIO that the PIR is connected to and ground. If you haven't 
already done it, a pull-up resistor between that pin and vcc can help as well

titomontes, 2017-08-29:
You can also create a metallic shield, and place the PIR as far away from the Wifi as possible. I used lead, it works very well.

fttx, 2017-08-29:
@davidelang Ok, il try with the pull-up resistor. But why put a capacitor between pir out and ground? The problem is that the pir sensor triggers when should not, and when it is already triggered the capacitor can't stop it in any way.

@titomontes lead? isn't it a bit overkill? 😆

davidelang, 2017-08-30:
On Tue, 29 Aug 2017, filippo wrote:

> But why put a capacitor between pir out and ground? The problem is that the pir sensor triggers when should not, and when it is already triggered the capacitor can't stop it in any way.

if there is some AC signal picked up by the wire (acting as an antenna), the 
capacitor gives it a path to ground while having little effect on the DC signal 
of the switch opening or closing. this makes it far less likely that stray 
signals will trigger a false reading.

fttx, 2017-08-30:
I think that the problem is before of the output pin, because it stays on for all the duration of the pir module timer, the interference is disturbing the pir circuit, not the output wire...

titomontes, 2017-09-03:
Off course it´s overkill. It´s also effective and cheap, easy to implement and solderless.

Cheers!

jetema, 2017-09-08:
fttx you can try covering the sonoff with one or more pieces of aluminium foil, that will decrease strenght os wifi signal ( on both directions ) and check if that trick improves the sensitivity of your PIR  ;-)

jhlweb, 2017-10-26:
Just use a piece of shielded FTP wire. or wrap the wires in aluminum foil and also wrap the PCB of the PIR in some paper tape first and then with aluminum foil. this should work. And make sure that all foil is connector to the ground of the circuit.

closed 2018-05-07 by stale[bot]