arendst/Tasmota #808: If MQTT SSL is used there is a stop required after fingerprint test

stefanbode, 2017-08-28:
Please insert in sonoff.ino after line 637 a stop() of the espclient. Otherwise the next mqtt.connect causes the second open connection creating interferences. The first socket will close after a while but MQTT needs at least one reconnect to overcome the problem. The connection is smooth and reliable it the client is stopped after the verify.
+ line 637 sonoff.ino espClient.stop();

```
    if (espClient.verify(sysCfg.mqtt_fingerprint, sysCfg.mqtt_host)) {
      addLog_P(LOG_LEVEL_INFO, PSTR("MQTT: Verified"));
    } else {
      addLog_P(LOG_LEVEL_DEBUG, PSTR("MQTT: Insecure connection due to invalid Fingerprint"));
    }
    espClient.stop();
    yield();
#endif  // USE_MQTT_TLS
```

liorsu, 2017-10-05:
Seems to me like it is lacking the stop as well.

stefanbode, 2017-10-05:
Hi @arendst would be great if you can commit this small change to the codebase.

arendst, 2017-10-09:
Will be in next (pre)release but for now my fingerprint doesn't work anymore and I forgot how to get a new one for my m20.cloudmqtt.com ...

closed 2018-05-07 by stale[bot]