arendst/Tasmota #809: Sonoff Dual, cannot set it to module 5

Dappie, 2017-08-28:
Hi,

Problems with the sonoff dual
First: Mine communicates just like a 'Sonoff Basic' at 115200 bps

I have welded a wire as described on the on the Tasmota github page, put it to ground and flashed the  Sonoff.  After that I can communicate with the device over wifi (console).

I tried to change the setting to Sonoff Dual by the command 'module 5'
The device shows that its set to Sonoff Dual, reboots en falls back to Sonoff basic.

I have read something about configuring trough WPS but don't understand that.

Any idea?

davidelang, 2017-08-28:
did you install the full version or the stripped down version that's only 
supposed to be used to bootstrap the full version? the stripped version won't 
save any settings

Dappie, 2017-08-28:
I installed the 'BE_MINIMAL' version, could that be the problem?

Dappie, 2017-08-28:
:-) :-) So simple (if you know) - that was the problem 
Thanks @davidelang

davidelang, 2017-08-28:
Yes, that version doesn't save any config settings, it's just to let you install 
the full version.

Dappie, 2017-08-28:
Again, thanks

closed 2017-08-28 by Dappie