arendst/Tasmota #810: Feature Requests

prahjister, 2017-08-29:
2 feature requests for the RF Bridge

When the bridge recieves any rf it sends something like the following to the topic

tele/sonoffrfbridge/RESULT {"RfReceived":{"Sync":11560, "Low":320, "High":1130, "Data":"982817", "RfKey":"None"}}

Is it possible to add an option to only send the int value of the data to the topic for example above would look like below. 
tele/sonoffrfbridge/RESULT 9971735

Second request 

Due to the one way nature of 433 mhz. Sensors usually send 10 or so signals and sometimes the rf bridge publishes duplicates. Could a timer be setup so duplicates can be avoided for a couple seconds.

prahjister, 2017-08-29:
Here is some code from https://github.com/1technophile/OpenMQTTGateway/blob/master/OpenMQTTGateway.ino to help. 

unsigned long ReceivedSignal[array_size][2] ={{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0},{0,0}};

boolean isAduplicate(long value){
trc(F("isAduplicate"));
// check if the value has been already sent during the last time_avoid_duplicate
for (int i = 0; i < array_size;i++){
 if (ReceivedSignal[i][0] == value){
      unsigned long now = millis();
      if (now - ReceivedSignal[i][1] < time_avoid_duplicate){ // change
      trc(F("---don't publish the code duplicate---"));
      return true;
    }
  }
}
return false;
}

arendst, 2017-08-29:
First feature won't be implemented but perhaps you could use command ``SetOption4 1`` to trigger your tool better (Changes common RESULT to something more specific).

Second option is feasible and will be implemented.

prahjister, 2017-08-31:
Thanks for the fix. I was able to get home assistant working. Here are some examples. It took me a minute to realize it was a nested json. Maybe add to your wiki

```
binary_sensor:
  - platform: mqtt
    name: "Living Room Motion"
    payload_on: "98281B"
    payload_off: "9971736"
    sensor_class: motion
    state_topic: "tele/sonoffrfbridge/RESULT"
    value_template: '{{ value_json.RfReceived.Data }}'

automation:
  - alias: Living Room Tamper
    hide_entity: true
    trigger:
      platform: mqtt
      topic: tele/sonoffrfbridge/RESULT
    condition:
      condition: template
      value_template: '{{ trigger.payload_json.RfReceived.Data == "982817" }}'      
    action:
      service: homeassistant.toggle
      entity_id: light.living_room
```

Tadzikj, 2017-09-14:
Hello
Can I add code command example (RfKey<x> 4 A401F1) save pilot code?

closed 2018-05-07 by stale[bot]