arendst/Tasmota #811: Request - TEMT6000 support

nightcat91, 2017-08-29:
Hey,
would it be possible to add support for this One-Wire Light Sensor?
This is the device:
https://www.aliexpress.com/item/Smart-Electronics-TEMT6000-An-ambient-light-sensor-Simulate-the-light-intensity-module-Visible-light-sensor/32583469115.html

There are already lots of sensor using the One-Wire Interface, so my guess is that it would only involve reading the data from this one correctly to display it.

Would be great, thanks

PeggyFree, 2017-08-29:
This is not a digital sensor, but an analog one. 

> The TEMT6000 is a silicon NPN phototransistor housed in a miniature
transparent mold. It provides a wide angle of up to ± 60 degrees of half sensitivity and provides
analog output relative to ambient light. 

In the product you linked I don't see any 1-Wire interface, only the TEMT6000.

nightcat91, 2017-08-29:
Thanks for the clarification.

Looking at the code I used so far, it also uses this as analog sensor.
The code I used so far can be found here:
https://github.com/bruhautomation/ESP-MQTT-JSON-Multisensor/blob/master/bruh_mqtt_multisensor_github/bruh_mqtt_multisensor_github.ino

At line 74 you can see that the LDR is assinged to Pin A0
And at lines 480-485 there is the code that reads a new value from the sensor if I get that right...

Can this be used with the Tasmota software?

ionciubotaru, 2017-08-29:
A NPN phototransistor is not un analog, it is a digital switch - use it as switch1, 2 3 or 4 and connect it to any digital pin available

nightcat91, 2017-09-13:
Hi,
sorry for the long time... I was on vacation.

@ionciubotaru I tried setting it up as a digital switch (Switch1) and it does work, but not the way I want it to work. I am getting "toggle" and "hold" commands over mqtt from Switch 1 now when I move the sensor between bright and dark areas. 

But I need an analog reading from the sensor to get the light level (in lumen, or lux)

The NodeMCU has a single pin for analog, as far as I can see form various images about the pins of that board. (Pin A0)

So i am still looking for a solution to get this sensor to work. 
Would be really great since its such a cheap and tiny sensor!

ionciubotaru, 2017-09-14:
According to sparkfun https://learn.sparkfun.com/tutorials/temt6000-ambient-light-sensor-hookup-guide TEMT6000 act like a photoresistor. You can you use a Tasmota's usual analog read.

From web interface select - CONFIGURATION -> CONFIGURE MODULE -> 18. WEMOS D1 mini
press SAVE
after esp resets go again to - CONFIGURATION -> CONFIGURE MODULE and select `ADC0   A0 Analog input` on analog pin.

nightcat91, 2017-09-14:
I looked at my module (NodeMCU) webpage and wasnt able to find the analog pin in "Configure Module", but it is set to "Wemos D1 mini".
Then I had a look at the release notes and searched for "Analog" and I found this line under V5.6
"* Add AnalogInput0 if configured as Analog Input to webpage (#697, #746)"
Since I was still using V5.5.1 on my NodeMCU I will update the Board once I get back home and report back afterwards.

Thanks for your help so far!

davidelang, 2017-09-14:
the analog pin needs to be enabled at compile time, by default (and in the 
precompiled binaries), it's measuring input voltage.

nightcat91, 2017-09-14:
Thanks for all the help, I was able to get the analog readings of the sensor to show up.

I was hoping for a nice display of a human readable value like with some of the other sensors. 
I might stick with the I2C BH1750 ambient light sensor...

Maybe someone will one day write the missing code for the TEMT6000 to be as easy to set up as the other sensors are right now.

Thanks again for all the people that helped me, I learned a lot!

dawsmac, 2017-10-01:
Hi, nightcat91, 

can you point me in the right direction I am trying to achieve this with my Sonoff Basic 👍 

Cheers

Rich

nightcat91, 2017-11-25:
Hello,

I am back to see if I could get some help in using the TEMPT6000 sensor.
Using V.5.9.1 I connected the sensor to the analog input and I got a reading shown on the webpage.

I would love to get this number to something useful, preferably Lux.
At first it seemed as this was already some sort of lux reading, but it turned out not to be that way.
I compared it to the BH1750 Sensor (I2C Illuminace Sensor).

At a low light condition the values where like this:
Analog: 27
BH1750: 21Lux

At high values those got very different:
Analog: 575
BH1750: 1336 Lux

Anyone got an idea how to get a Lux value from the Analog value?
Thanks

closed 2018-05-07 by stale[bot]