arendst/Tasmota #813: GPIO0 VS GPIO3 ( WallSwitch)

dawsmac, 2017-08-29:
Hi Guys, 

I have flashed the Sonoff TH 10 and wired a switch between GPIO3 and ground with the 10k resistor.  
from the UI, I have selected "09 Switch 1" and also fire cmd switchmode1 0 and disabled logging. Looking in the console I can see the extrnal switch sending  tele/HW/SENSOR = {"Time":"2017-08-29T12:48:58", "**_Switch1":"ON"_**, "AM2301":{"Temperature":25.5, "Humidity":58.9}, "TempUnit":"C"}

but not firing the relay, should use GPIO0 or have I missed something simple ;) 

Cheers

Rich

ionciubotaru, 2017-08-29:
set `switchTopic 0` , than switch1 will fire relay1

dawsmac, 2017-08-30:
Hi ionciubotaru, 

I have tryied the switchTopic but getting the following.. 

tele/HW/STATE = {"Time":"2017-08-30T14:20:38", "Uptime":16, "Vcc":3.204, "**POWER":"OFF**", "Wifi":{"AP":1, "SSID":"**********", "RSSI":66, "APMac":""}}
14:20:38 MQTT: **tele/HW/SENSOR** = {"Time":"2017-08-30T14:20:38", "**Switch1":"ON",** "AM2301":{"Temperature":23.7, "Humidity":53.5}, "TempUnit":"C"}

ionciubotaru, 2017-08-30:
send me the answer from `switchTopic` command
then try `switchmode 1` open and close the switch, then try `switchmode 2` and so on till ``switchmode 4`

In one of this modes the relay must change its state (i think in mode 1 and 2)
switchtopic must be 0

dawsmac, 2017-08-30:
CMND: switchTopic 0
MQTT: stat/HW/RESULT = {"SwitchTopic":""}

Going to try the switchmode's now :)

dawsmac, 2017-08-30:
Am I missing something really simple, so switchmode 1/2 both seem to toggle the relay. 
but let say I power cycle the sonoff or press the button on the sonoff when the sensor ( pickup switch1 is ON does not toggle the relay ;/ 

7:59:42 MQTT: tele/Heating/STATE = {"Time":"2017-08-30T17:59:42", "Uptime":1, "Vcc":3.204, "POWER":"ON", "Wifi":{"AP":1, "SSID":"WIFI", "RSSI":66, "APMac":"*************"}}
17:59:42 MQTT: tele/Heating/SENSOR = {"Time":"2017-08-30T17:59:42", "Switch1":"ON", "AM2301":{"Temperature":23.9, "Humidity":53.7}, "TempUnit":"C"}
17:59:54 MQTT: stat/Heating/RESULT = {"POWER":"OFF"}
17:59:54 MQTT: stat/Heating/POWER = OFF
18:00:32 MQTT: tele/Heating/STATE = {"Time":"2017-08-30T18:00:32", "Uptime":1, "Vcc":3.205, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"WIfi", "RSSI":68, "APMac":"**************"}}
18:00:32 MQTT: tele/Heating/SENSOR = {"Time":"2017-08-30T18:00:32", "Switch1":"ON", "AM2301":{"Temperature":23.9, "Humidity":54.1}, "TempUnit":"C"}

dawsmac, 2017-08-30:
OK, it might be working correctly and i might have the wrong understanding of the external switch.
Let me know if what i am thinking is right or wrong.

if I wanted to have an external Switch (Master) when switched to ON fires the relay to ON at same time override the software switch. Even if there was a powercut etc... it would come back in the ON status ( relay )  

I have it set to switchmode 0 and switchtopic 0

ionciubotaru, 2017-08-30:
if you set switchtopic to 0 every time you change the switch it change the relay state
in the same time you can change the relay state via mqtt

try all 4 swithmodes to see if one of them feet your needs
If not you have to wait for [switchmode 5 implementation](https://github.com/arendst/Sonoff-Tasmota/issues/801)

dawsmac, 2017-09-05:
Thanks for all you help Guys :)

closed 2017-09-05 by dawsmac