arendst/Tasmota #814: Multiple DS18B20

Dickesplumpel, 2017-08-29:
Hello!
I'm a great fan of your firmware and using it in every of my ESP8266s - 01, 07, Nodemcu, Wemos and Sonoff. I have only one issue i can't solve (i have had many but always found a solution ;-)):
In one projekt i have 4 ds18b20 and one switch. Everything worked fine for weeks till one day the connection to 2 ds18b20 breaks sometimes and the numbers DS1, DS2 etc. change because they are not fixed to one sensoraddress. I have reduced the numbers of sensors to 2 with the same effect and i also have changed the ESP from 8266-01 to Wemos-board with no better result, i have changed the sensors, powered up with 5v Vdd and 3,3v over Data line with 4,7k resistor, changed the firmware version - nothing helped that sometimes some sensors are not "readable", the cables are max. 8m long self twisted phonecables (dickere Telefonkabel).
It's not urgent, has someone had the same problem ans fixed it?
Thanks a lot, Dickesplumpel

ionciubotaru, 2017-08-29:
I had the same problem, it is a hardware, not a software one. Put all 4 DS's near ESP and you will see that all 4 appears on the web interface. If the cable is 8m long only few are detected. 

In this case:
- use 5V to supply power to DS. (put 1K resistor between data cable and ESP pin to proctect ESP against 5V)
- use a shielded cable (SFTP or better)
- put the GND and data on the same pair

Finally if nothing works, try to move ESP near DS's (in my case this was the solution)

Dickesplumpel, 2017-08-29:
Thanks for your reply!
I know that it is a hardware problem but probably the temperature value
could be stored as long as there is no new value for this DS or that the
number of the sensor could be bound to the address of it. In my case it is
absolutely enough when the temperature is read one time in 1-2 minutes. I
have 5V power supply and 3,3v on data cable with 4,7k resistor between.
Isnt 5V too much?
Shielded sftp cable is a good idea, i have some meters cat7 to test. ESP
near DS doesn't work because all DS are in different directions.
If nothing works i give every sensor an 8266-01 but thats no challenge ;-)
My english is a bit rusty :-)))

ionciubotaru, 2017-08-29:
I don't know how much memory is available to store sensor address, temperature, last time the sensor was seen, all multiply by 8 (the maximum DS's). Only @arendst can decide howto use the available memory

Dickesplumpel, 2017-08-29:
Thats right and good because i am no programmer - written some stuff in C
in the late 80's - nothing left of in my brain and iam very glad that Theo
has written the stuff i needed to play with. A look in the ds18x20.ino made
me think of this option but i have no idea where to put the fixed stuff -
its pretty flexible. I can read the address send by mqtt in home assistant
the same way as the temperature but its send in the bracket of the ds
number, also shown in the tele/sonoff/SENSOR
I think the cable with gnd and data in one pair is the solution - ill try
it out this week.
Thank you for your help!

Dickesplumpel, 2017-09-01:
Changed Cable, changed sensors, changed firmware, changed delay time,
changed ...changed power adaptor - bang - stable! 4 sensors with 2 x 4m,
one 8m and one 10m self twisted
 phone cable.

2017-08-29 21:17 GMT+02:00 Armin Krämer <armin.kraemer@web.de>:

ionciubotaru, 2017-09-01:
@Dickesplumpel please published detailed instructions about your work
maybe is useful for other users

Dickesplumpel, 2017-09-01:
More than 10 years old power supply GX34W-5-12 for external HDs
5V to sensors DS18B20 and power converter AMS1117 for ESP8266-01, firmware
5.5.2 - (12V of power supply left open)
6,3V 1000µF condenser between 3,3V and GND, 4,7k between GPIO2 and sensor
data
sensor cables self twisted, 3 cables in elektric drill and twist,
approximatly 2-3 times per centimeter, thicker phone cable approx. d=0,6mm
Everything together on a breadboard, maybe better when sold (gelötet), try
it in autumn or winter or sooner when stuck again ;-)
If the pictures can be seen: GPIO1 (TXD) controls if a waterpump (Grundfoss
GENI) runs or not

Dickesplumpel, 2017-09-01:
I think its only 1-2 times twisted per cm.

2017-09-01 20:26 GMT+02:00 Armin Krämer <armin.kraemer@web.de>:

CrazyIvan359, 2017-10-12:
If it helps any, I have a device with 4 DS18B20s and 3 DHT22s on it. The sensor data packet was really long, so I condensed the presentation of the DSB sensors.

Around line 200 in xsns_ds18x20.ino there is the following code:
`snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s%s\"DS%d\":{\"" D_TYPE "\":\"%s\", \"" D_ADDRESS "\":\"%s\", \"" D_TEMPERATURE "\":%s}"), mqtt_data, stemp1, i +1, dsbstype, ds18x20_address(i).c_str(), stemp2);`
Which I replaced with this:
`snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s%s\"%s\":%s"), mqtt_data, stemp1, ds18x20_address(i).c_str(), stemp2);`
So that the DSB portion of the sensor data looks like this:
`"DS18x20":{"28FF1618A316033A":36.8, "28FFC367A3160384":34.4, "28FF47F5A21605C0":28.7, "28FFA75DA31603D9":28.6}`
That way I can just extract the temperature reading from the sensor I want by doing
`$.DS18x20.28FF1618A316033A`

Dickesplumpel, 2017-10-12:
Thank you very much for your reply, i'll try it out next time i jump into
playing with IOT's - in the moment everything works fine!

Dylantje, 2018-09-15:
Dear All

Please some help to read 6-8 dallas into domoticz?
I have 8 dallas sensors on my sonoff basic i see them in the webinterface.
And i can set 1 sensor with the idx  to send to domoticz
Only how do i read the other 7??
 I heard something about rules..
Only that part i do not understand or how this part is working?
So please some one?

Thanks
[image]
[image]