arendst/Tasmota #815: Wemos D1 Mini Switch mode MQTT result

gheesung, 2017-08-30:
Hi

I am assigning a switch to GPIO4 at Wemos, set the `switchmode1 2` and `switchtopic 1`. The switch is able to switch on/off but I am getting "Unknown Command". Any idea what is happening? Thanks.

`09:37:16 CMND: switchmode1 2
09:37:16 MQTT: brownstone/bedroom/switch/stat/RESULT = {"SwitchMode1":2}
09:38:03 CMND: SwitchTopic 1
09:38:03 MQTT: brownstone/bedroom/switch/stat/RESULT = {"SwitchTopic":"switch"}
09:38:17 MQTT: brownstone/bedroom/switch/cmnd/POWER1 = OFF (retained)
09:38:17 MQTT: brownstone/bedroom/switch/stat/RESULT = {"Command":"Unknown"}
09:38:19 MQTT: brownstone/bedroom/switch/cmnd/POWER1 = ON (retained)
09:38:19 MQTT: brownstone/bedroom/switch/stat/RESULT = {"Command":"Unknown"}`

ionciubotaru, 2017-08-30:
`switchTopic` works after you define switch1, 2,3 or 4

gheesung, 2017-08-30:
Hi

My switch1 is already defined but I do not have switch2,3,4 . Switch1 is able to trigger on/off remote switch. 

I tried to define everything but still having the "unknown" command problem. Am I doing it wrongly?

12:30:34 CMND: SwitchMode 2
12:30:34 MQTT: brownstone/bedroom/switch/stat/RESULT = {"SwitchMode1":2}
12:30:47 CMND: switchmode2 2
12:30:47 MQTT: brownstone/bedroom/switch/stat/RESULT = {"SwitchMode2":2}
12:30:55 CMND: switchmode3 2
12:30:55 MQTT: brownstone/bedroom/switch/stat/RESULT = {"SwitchMode3":2}
12:31:04 CMND: switchmode4 2
12:31:04 MQTT: brownstone/bedroom/switch/stat/RESULT = {"SwitchMode4":2}
12:31:07 MQTT: brownstone/bedroom/switch/cmnd/POWER1 = ON (retained)
12:31:07 MQTT: brownstone/bedroom/switch/stat/RESULT = {"Command":"Unknown"}
12:31:09 MQTT: brownstone/bedroom/switch/cmnd/POWER1 = OFF (retained)

ionciubotaru, 2017-08-30:
One switch is enough. Maximum number of switches is 4.
Try to set `switchTopic`. If you attach switch1 to gpio14 (for exemple) you have to set `switchmode1 <yourvalue>`, `switchtopic1 <yourvalue>`

gheesung, 2017-08-30:
I think I have found the symptom and workaround but do not know the root cause. I am using a wemos d1 mini as a switch to turn on/off another Sonoff RF. The remote Sonoff is behaving normally and the "unknown" command is received by d1 mini. The following log is captured by syslog at raspberrypi.

Aug 30 20:14:19 **switch-0235** ESP-MQTT: brownstone/bedroom/switch/cmnd/POWER1 = ON (retained)
Aug 30 20:14:19 **switch-0235** ESP-MQTT: brownstone/bedroom/switch/stat/RESULT = {"Command":"Unknown"}
Aug 30 20:14:19 **sonoff-rf** ESP-MQTT: brownstone/bedroom/switch/stat/RESULT = {"POWER":"ON"}
Aug 30 20:14:19 **sonoff-rf** ESP-MQTT: brownstone/bedroom/switch/stat/POWER = ON
Aug 30 20:14:23 **switch-0235** ESP-MQTT: brownstone/bedroom/switch/cmnd/POWER1 = OFF (retained)
Aug 30 20:14:23 **switch-0235** ESP-MQTT: brownstone/bedroom/switch/stat/RESULT = {"Command":"Unknown"}
Aug 30 20:14:23 **sonoff-rf ESP**-MQTT: brownstone/bedroom/switch/stat/RESULT = {"POWER":"OFF"}
Aug 30 20:14:23 **sonoff-rf ESP**-MQTT: brownstone/bedroom/switch/stat/POWER = OFF

**Then I configured a dummy relay to one of the GPIO pin in D1 mini, Sonoff RF received the correct result**
Aug 30 20:19:32 switch-0235 ESP-MQTT: brownstone/bedroom/switch/cmnd/POWER1 = ON (retained)
Aug 30 20:19:32 sonoff-rf ESP-MQTT: brownstone/bedroom/switch/stat/RESULT = {"POWER":"ON"}
Aug 30 20:19:32 sonoff-rf ESP-MQTT: brownstone/bedroom/switch/stat/POWER = ON
Aug 30 20:19:32 switch-0235 ESP-MQTT: brownstone/bedroom/switch/stat/RESULT = {"POWER":"ON"}
Aug 30 20:19:32 switch-0235 ESP-MQTT: brownstone/bedroom/switch/stat/POWER = ON (retained)
Aug 30 20:19:33 switch-0235 ESP-MQTT: brownstone/bedroom/switch/cmnd/POWER1 = OFF (retained)
Aug 30 20:19:33 sonoff-rf ESP-MQTT: brownstone/bedroom/switch/stat/RESULT = {"POWER":"OFF"}
Aug 30 20:19:33 sonoff-rf ESP-MQTT: brownstone/bedroom/switch/stat/POWER = OFF
Aug 30 20:19:33 switch-0235 ESP-MQTT: brownstone/bedroom/switch/stat/RESULT = {"POWER":"OFF"}
Aug 30 20:19:33 switch-0235 ESP-MQTT: brownstone/bedroom/switch/stat/POWER = OFF (retained)

lgnd33, 2017-08-30:
The way it works is following: 
* `switchtopic 1`, without defining a switchtopic, will publish a message into `"cmnd/<own_name>/"`
* Tasmota reacts on `"cmnd/<own_name>/"` commands and will try to do something whenever it gets one from MQTT. 

Not setting a separate topic is resulting in Tasmota publishing a command into it's own channel, confusing itself and resulting in `"stat/<own_name>/RESULT = command: unknown"`. This, of course, happens in your setup with Wemos, where you have buttons (generating command), but not relays that are mapped to it. 
Normally, with regular Sonoff devices, every button is mapped to a relay (BUTTON1 -> RELAY1), so press of a button will generate a command into a cmnd channel, that gets executed by assigned relay. `stat/<own_name>/RESULT` just confirms the execution of a command. This is why your "dummy relay" trick worked. 

What you *probably* want to do is following: 
`
switchtopic "bedroom_buttons"
switchtopic 1
`
and then monitor `cmnd/bedroom_buttons/#` for changes from some home automation software. Depending on your button configuration, you might get `cmnd/bedroom_buttons/POWER1 = TOGGLE` or similar messages.

gheesung, 2017-09-01:
Thanks for the tips.

closed 2017-09-03 by gheesung

roblad, 2017-11-29:
Hi,

I do have the same problem, but realy I do not know what exactly comands should I execute on console, described "What you probably want to do is following:
switchtopic "bedroom_buttons" switchtopic 1"

Does not work, could you give me the step by step actions that I should perform to get that result. I do not have RELAY1 and 2 defined, But I wudlike to se result of BUTTON/SWITCH when I pussh them - on MQTT and console, could you help me ?