arendst/Tasmota #816: I got problem for upload firmware to Huafan SS

hamwong, 2017-08-30:
I tried upload firmware with Arduino IDE, espytool, all result like below, Huafan fail to start,
I have tried esptool.py --port COM5 erase_flash, I also tried github standard firmware all result same

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�,5�!{x�.搦躩E殥��(�+z
7颸�*�(鎴�)雱�9H珶p郵扢~躩�(晡ro|�-筋�,���a頛係)z�(~
`
!谽
|Z~吩婢遰{|屔H,徙指,�硝�a頛係)P�*�-�$�+PPrJv,�硝�a頛弈)z�(~
`)谽oP齹l�4r泠y�-l敔� ,�5B�a頛弈)P�(Y*�-�$搢�+PPrJv,�湎；e�+�$t眠*~`z渽5

PeteBa, 2017-08-30:
Hamwong, just to check, did you set the Crystal Frequency to 40MHz in the Arduino IDE Tools menu? This is different to the default setting of 26MHz. There are some notes in the Wiki specifically for the Huafan device that shows how to add this setting if it isnt already showing in the menu [here](https://github.com/arendst/Sonoff-Tasmota/wiki/HuaFan-Smart-Socket)

PeteBa, 2017-08-30:
Great.

Just to rule out another problem, is your terminal window set to 115200 baud? Reason I ask is that the output you posted above has the bootlog ( i.e. first few lines up to ld) clear but what follows is garbled.  Generally the initial boot log from the esp8266 is sent at 76800 and then the Tasmota firmware changes the baudrate to 115200. So generally the terminal window is set to 115200 and you get garbled bootlog but clear tasmota log.  Whereas this looks the other way around.

hamwong, 2017-08-30:
you are right with the speed, I dont know will this issue due to upload speed?
I got no issue with other esp product and sonoff product after upload frimware,
but this one, it fail to start up~

PeteBa, 2017-08-30:
I have 8 HuaFan switches with Tasmota firmware in the house and have only had problems flashing due to incorrect settings (i.e. Crystal Frequency) or insufficient power supply during flashing process.

A few things to help understand:
1) Are you tapping into the esp8266 daughter board or are you using the "solder free" method and connecting to the main board for most pins?
2) Can you take a screen shot of the setting in the Tool menu and post here? We can confirm that the settings match.
2) Can you open the serial monitor window in the Arduino IDE with baudrate set to 115200 and capture the Huafan output during a restart of the switch? Does it show any output logged?

The only other issue I have had with the Huafan is providing enough power to ensure a stable flash of firmware. How are you powering the Huafan during the upload process? Are you just using the FTDI or do you also have an independent 5V/3.3V power supply connected. The FTDI cannot supply enough current to power the device properly.

hamwong, 2017-08-31:
Thanks for your help,
1)I am using "solder free"
2)[url=https://imgbb.com/][img]https://image.ibb.co/mPD9HQ/1.jpg[/img][/url]
3)[url=https://ibb.co/ixQpHQ][img]https://preview.ibb.co/nC3bxQ/2.jpg[/img][/url]

I got all result same as 3) with difference upload way, no matter by Arduino 1.8 with  baudrate set to 115200 or 57600, than I tried with  esptool, also fail,

I got independent 5V, which is exactly same as no pin pic.

I think if I got issue, may be only on ESP8266/Arduino code 2.4.0-rc1, as I only copy Sonoff-Tasmota/arduino/version 2.4.0-rc1/boards.txt & platform.txt into arduino-1.8.0\portable\packages\esp8266\hardware\esp8266\2.3.0 and add 3 line crystal into boards.txt

PeteBa, 2017-08-31:
OK, it does look like the crystal frequency. I have just replicated this on my machine. So to fix this I suggest the following:
1) Flash Mode should be DOUT
2) You need the latest ESP8266 core as follows:
      - Start Arduino and open Preferences window.
      - Enter https://github.com/esp8266/Arduino/releases/download/2.4.0-rc1/package_esp8266com_index.json into Additional Board Manager URLs field. You can add multiple URLs, separating them with commas.
      - Open Boards Manager from Tools > Board menu and install esp8266 platform (check that it is using the RC version).
3) Rebuild and flash

If that fails then can try flashing the firmware attached. It is the latest master branch with default settings and RC version of the ESP8266 core. [huafan.ino.zip](https://github.com/arendst/Sonoff-Tasmota/files/1266896/huafan.ino.zip)

Let me know if you have success.

hamwong, 2017-08-31:
Great Thanks, I will try it once I back home

hamwong, 2017-08-31:
Great Thanks, It works, I do as you said and It works perfectly, Great Thanks,
I like this socket very much but original firmware is rubbish, now I can buy more as finally I can use it on this favorite firmware

closed 2017-08-31 by hamwong