arendst/Tasmota #818: Sonoff TH as thermostat

HighMasterGogo, 2017-08-31:
Please excuse my noob-ish-ness with all of this! I'm looking for some advice, please.

I've recently flashed all my basic Sonoffs with the Tasmota firmware and it's working great. Awesome firmware.

Now, I'm looking to setup my Sonoff TH16 with AM2301 temp/humidty sensor to work with Homebridge as a thermostat for my boiler.

I'd very much like to incorporate the Tasmota firmware as part of this build but I'm not sure if it's possible?

If anyone could provide any advice I'd be very grateful.

davidelang, 2017-08-31:
The tasmota firmware does not include thermostat functionality (it's a 
frequently requested feature), but it can be connected to a sensor and report 
that sensor output to your home automation software, and that software can then 
turn things on and off.

HighMasterGogo, 2017-09-01:
Thanks, David. I appreciate the quick reply.

I see that Tasmota is frequently developed; if thermostat is a heavily requested feature, do you think there's a good chance it will be developed soon?

davidelang, 2017-09-01:
search the issues for thermostat and read the other issues on the topic.

IceStuff, 2018-04-08:
Hello All,

I have extended an existing project called Hard Thermostat with webpage controls, heating and cooling modes and supports all temperature sensors.

Check it out and provide feedback would be great.  Humidity control is coming.

https://github.com/IceStuff/Sonoff-Tasmota/releases/tag/v5.12.1

Thanks,
IceStuff

ascillato, 2018-05-21:
Hi,

In the wiki, now is available an example of a Simple Thermostat using RULE feature (https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#7-simple-thermostat-example)

Please, check it to see if that helps you.

If this solves your issue, please close it. Thanks +1

closed 2018-06-13 by ascillato2

Makk1983, 2018-11-21:
Hi, sorry for the translation with google, I'm new, I tried the firmware tasmota with the integrated thermostat of IceStuff, for what I have to do it works well enough, you know what are the commands web and mqtt to be able to command or query via http command or via mqtt broker? Thank you so much

ascillato, 2018-11-21:
Hi,

If you are using IceStuff fork, may be it is better to ask directly to him, I don't know what commands or modifications he have.

If you need support with the Original Tasmota, you can use the Tasmota Chat at Discord (https://discord.gg/Ks2Kzd4)
If you need information about Commands, you can find that in the wiki: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.