arendst/Tasmota #819: AP(Access Point) mode password protection

EmilySmith0, 2017-08-31:
I have **Sonoff 4CH** relays working normally since a **month**. Today I **switched off** my WiFi router to change location. I saw all Sonoff blue lights were blinking due to WiFi unavailability. All were in **AP mode**. I connected my Laptop with one of the relays without any password. Now my SSID and password can been seen in Chrome as I **removed password dots**.

Then I turned on my WiFi router and that **4CH was not able to connect** with my router until I pressed button.

I think there is a big issue without access point password protection.

Kindly consider adding password in next release

Someone please guide me how can I add AP mode Password.
Thanks 👍

ionciubotaru, 2017-08-31:
1. Use `WebPassword` to protect your web interface
2. Disable AP mode, use only wifi client in infinite retry mode
3. disable web server al all

EmilySmith0, 2017-09-01:
I know its one way to deal with it.
user_config.h Line # 46
#define WIFI_CONFIG_TOOL       WIFI_RETRY        // [WifiConfig] Default tool if wifi fails to connect
But when Configuring WiFi pressing GPIO0 four times makes module in AP mode. After successfully connection with WiFi WIFI_CONFIG_TOOL changes its value to WIFI_WPSCONFIG.
Now I have to use command  WifiConfig 4 to make it in retry mode again.
But its no more plug and play module as ESPURNA is
Tasmota is way better than ESPURNA in all aspects except admin user/Password and AP mode Password protection

I was looking through ESPEASY MEGA and ESPURNA for AP point protection and was not able to understand
Thanks

ionciubotaru, 2017-09-01:
`SetOption1 1` Allow only single, double and hold press button actions
In this way no AP any more. Try to configure your device via mqtt only - this is the most versatile way
Web interface is just for tests, it is recommended to disable it in production mode

EmilySmith0, 2017-09-01:
Thanks @ionciubotaru I have already applied this SetOption1 to 1 and web interface is enabled as it is password protected.

Its a reliable solution.

I am not closing issue now. May be @arendst considers AP Password protection.
I will close it within next week.
Thanks again
Emily

lobocobra, 2017-09-03:
You should close it.... because this is not a coding problem, but a configuration problem from your side. 
=> if it is closed, it is still around to help newcomers like me

I just learned about SetOption1... :)

wojoco, 2017-09-03:
@ionciubotaru  please ,where can I find  "setoption1" ?  user_config.h?

EmilySmith0, 2017-09-03:
In settings.ino
line number 418
`//  sysCfg.flag.button_restrict = 0;`

change it with 
`sysCfg.flag.button_restrict = 1;`

ionciubotaru, 2017-09-04:
You don't have to change any line inside code.
Read https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#setoption-overview

wojoco, 2017-09-04:
@EmilySmith0  @ionciubotaru    thanks!

closed 2017-09-07 by EmilySmith0

gymolnargabor, 2018-05-26:
I would like to have the same thing, password protect to the ap mode.