arendst/Tasmota #820: Toggle between two relays

LawaRobert, 2017-08-31:
Hi @ all, first I want to thank all of you for the fantastic work!

It is possible to toggle between two relays? One of the relays should always be on, e.g.
Relay 1 ON / Relay 2 OFF
Relay 2 OFF / Relay 1 ON

I tried different things and combinations, for exampe i used the comand SETPOINT 14 1 or configure the module with GPIO of the first relay with RELAY1 and GPIO of the second relay with RELAY1I. But I cannot find the right configuration for my intent.

davidelang, 2017-08-31:
no, there is not a way of forcing one relay to always be on.

However, there is an option to prevent two relays from both being on at the same 
time.

search for the 'shutter' option.

David Lang

LawaRobert, 2017-08-31:
With the command "SetOption14 1"  you can set an interlock mode for all relays. It should be similar to the shutter option.

ionciubotaru, 2017-09-01:
@davidelang please provide a link for shutter option - I didn't find it in wiki - supported commands

bapt1080, 2017-09-01:
> With the command "SetOption14 1" you can set an interlock mode for all relays. It should be similar to the shutter option.

shutter option => SetOption14 1 (InterLock Mode)

ionciubotaru, 2017-09-01:
Ok, I thought that shutter option is a different one.

lobocobra, 2017-09-02:
Out of the box, this is currently not possible. But check my "issue" 824#. I had a similar problem and while solving it, I did (by mistake) just generate that behavior, that always one relay is on. But luckily I could fix it, because it was a BUG and not a FEATURE for me.

You can do that by re-programming in the tab SONOFF the procedure "void do_cmnd_power(byte device, byte state)". You will want to look down 2/3 in the code of the procedure and specially have a look at  do_cmnd_power(i +1, 0);

With this code you can turn ON:    do_cmnd_power(i +1, 0);
and OFF:                                        do_cmnd_power(i +1, 0);
Any relay and you can add easily an "if condition" to turn relay  1 ON, if all others are off, respectively prevent that the last relay is turned off by a 2nd click.

The variable "device" contains the value that is to be changed, so you probably just want to check if it is already ON and prevent the OFF. You will need to compare the "device" value with the "imask" and "power" (hey that above tipp costed me 6 hours of searching, as I am not familiar with Arduino coding).

LawaRobert, 2017-09-03:
@lobocobra OK, thanks, but I realized it directly on the hardware yesterday.

I removed one of the n-channel (enhancement) MOSFET and placed an n-channel (depletion) MOSFET parallel to the remaining MOSFET (same GPIO).

closed 2017-09-03 by LawaRobert

Sunrise17, 2017-12-21:
Shutter mode (with interlock option) is available in Sonoff Touch with 2 gang?

zibous, 2017-12-21:
Hi,

Yes testet today:
With the command "SetOption14 1" you can set an interlock mode for all relays:
 Goto Console and enter the command: **SetOption14 1**

valentapat, 2021-06-21:
My setup - controllin relay 2 / relay 3

rule1 on power2#state=1 do Backlog rule2 0; power3 0; rule2 1 endon on power2#state=0 do Backlog rule2 0; power3 1; rule2 1 endon
rule2 on power3#state=1 do Backlog rule1 0; power2 0; rule1 1 endon on power3#state=0 do Backlog rule1 0; power2 1; rule1 1 endon
rule1 1
rule2 1