arendst/Tasmota #821: OTA flash reset configuration?

bapt1080, 2017-09-01:
Hello, first of all, thank you very much for this incredible project, I only use it on all my Esp8266 devices!
I wanted to know if it was possible to do an OTA update or via the web interface while keeping the module configuration?
Because I need to reconfigure everything every time I do an update (sonoff classic by default etc ...)

davidelang, 2017-09-01:
all the config settings should be retained by default when you upgrade. One of 
the more common questions we have here is why the upgrade didn't change settings 
that were set differently in the source.

ionciubotaru, 2017-09-02:
To keep the same configuration do not change:
`#define CFG_HOLDER             0x20161209        // [Reset 1] Change this value to load following default configuration parameters`

stefanbode, 2017-09-03:
I agree in theory this should work. And I can agree that it will work if you stay on the same version. Anyhow, if you do an OTO upgrade and change the version everything could be possible. I have seen all kinds of behavior. Sometimes it reboots, sometimes not. Sometimes it did not start up anymore, Sometimes it starts up but the SSID information is that corrupted that it did not connect. Sometimes some or all of the settings are corrupted. I know to do changes and allow backward compatibility is tough to reach. Especially if you have no memory. Any Dump read has the ability to destroy the rest if you change the configuration. arendst already try to avoid the nastiest one from the past now having already spare values defined he can use later on. But this does not solve all cases. I would add a duplicate of the config holder at the end of the dump file and check on reading if both matches because any structural change changes the length of the dump and this can be detected and if so maybe started with a clean configuration instead of hanging.

thayoz, 2017-09-19:
I'm also having the same issue, after an OTA update the Module parameters (I'm using GPIO14) to connect an external switch will be set back to **00 None**.

closed 2018-05-07 by stale[bot]