arendst/Tasmota #824: CH4 // Split Interlock mode of 4 channel in 2 groups

lobocobra, 2017-09-02:
Hi all

I wanted to use two 4CH Pro to control centrally 4 rollos. In consequence I needed to split the interlock mode of a 4CH/pro into 2 groups of 2 channels, in order to prevent, that the up/down motors can be activated the same time. 

If somebody is interested in the quick an dirty solution, you find it below. I tested it and it works like a charm. I combine it with PulseTime, which prevents, that the motors run by mistake forever.

You will need to add one variable and exchange to procedures in the code. See my changes in the answer sections. As it is quick an dirty, I doubt that it will make the way into the official code. But if there is interest, I am willing to invest more time.

lobocobra, 2017-09-02:
**As promised, find here the solution:**

**In user_config_overide.h add the following:**
#define Sonoff4CH_Interlock_SPLIT 0

**In sonoff, replace the procedure setRelay:**
void setRelay(uint8_t rpower)
{
  uint8_t state;

  if (4 == sysCfg.poweronstate) {  // All on and stay on
    power = (1 << Maxdevice) -1;
    rpower = power;
  }

 #ifdef Sonoff4CH_Interlock_SPLIT 0 // split your channels into 2 groups, each group allows one active channel only
 // ---- changes by lobocobra ----
 if (sysCfg.flag.interlock) {     // Allow only one or no relay set
    uint8_t mask = 0x01;
    uint8_t count = 0;
    byte result1 =0;
    byte result2 =0;
    for (byte i = 0; i < Maxdevice; i++) {
      if (rpower & mask) {
        if (i <2) { result1++;}//increment if low part is ON
        if (i >1) { result2++;}//increment if high part is ON 
      }
      mask <<= 1;
    }

    if ((result1) >1 && (result2 >1)) {power = 0; rpower = 0;}
    if ((result1) >1 && (result2 <2)) {power = power & 0x0C; rpower = power;} // 1/2 are both on and 3/4 max one is on
    if ((result1) <2 && (result2 >1)) {power = power & 0x03; rpower = power;} // 1/2 max one is on and 3/4 both are on     
       
  } 
 #else // use original code, only one channel allowed
 // ---- changes by lobocobra
   if (sysCfg.flag.interlock) {     // Allow only one or no relay set
    uint8_t mask = 0x01;
    uint8_t count = 0;
    for (byte i = 0; i < Maxdevice; i++) {
      if (rpower & mask) {
        count++;
      }
      mask <<= 1;
    }
    if (count > 1) {
      power = 0;
      rpower = 0;
    }
  }  
#endif
// --- end of modification by lobocobra

**In sonoof allso replace the procedure do_cmnd_power with my code**
void do_cmnd_power(byte device, byte state)
{
// device  = Relay number 1 and up
// state 0 = Relay Off
// state 1 = Relay On (turn off after sysCfg.pulsetime * 100 mSec if enabled)
// state 2 = Toggle relay
// state 3 = Blink relay
// state 4 = Stop blinking relay
// state 6 = Relay Off and no publishPowerState
// state 7 = Relay On and no publishPowerState
// state 9 = Show power state

  uint8_t publishPower = 1;
  if ((6 == state) || (7 == state)) {
    state &= 1;
    publishPower = 0;
  }
  if ((device < 1) || (device > Maxdevice)) {
    device = 1;
  }
  byte mask = 0x01 << (device -1);
  pulse_timer[(device -1)&3] = 0;
  if (state <= 2) {
    if ((blink_mask & mask)) {
      blink_mask &= (0xFF ^ mask);  // Clear device mask
      mqtt_publishPowerBlinkState(device);
    }

#ifdef Sonoff4CH_Interlock_SPLIT 0 // split your channels into 2 groups, each group allows one active channel only
// ---- modifications  by lobocobra ---- 

// find out if channel 1/2 or 3/4 are to be changed
if (device < 3 ) { // channel 1/2 are now changed
  // we are on 1/2
    if (sysCfg.flag.interlock && !interlockmutex) {  // Clear all but masked relay
      interlockmutex = 1;
      for (byte i = 0; i < 2; i++) {
        byte imask = 0x01 << i;
        if ((power & imask) && (mask != imask)) {
          do_cmnd_power(i +1, 0);
        }
      }
      interlockmutex = 0;
    }

} else {
  // channel 3/4 are changed
    if (sysCfg.flag.interlock && !interlockmutex) {  // Clear all but masked relay
      interlockmutex = 1;
      for (byte i = 2; i < Maxdevice; i++) {
        byte imask = 0x01 << i;
        if ((power & imask) && (mask != imask)) {
          do_cmnd_power(i +1, 0);
        }
      }
      interlockmutex = 0;
    } 
  
}
#else // original code where only 1 active channel is allowed
    if (sysCfg.flag.interlock && !interlockmutex) {  // Clear all but masked relay
      interlockmutex = 1;
      for (byte i = 0; i < Maxdevice; i++) {
        byte imask = 0x01 << i;
        if ((power & imask) && (mask != imask)) {
          do_cmnd_power(i +1, 0);
        }
      }
      interlockmutex = 0;
    }
#endif //end of modification of lobocobra

    switch (state) {
    case 0: { // Off
      power &= (0xFF ^ mask);
      break; }
    case 1: // On
      power |= mask;
      break;
    case 2: // Toggle
      power ^= mask;
    }
    setRelay(power);
#ifdef USE_DOMOTICZ
    domoticz_updatePowerState(device);
#endif  // USE_DOMOTICZ
    pulse_timer[(device -1)&3] = (power & mask) ? sysCfg.pulsetime[(device -1)&3] : 0;
  }
  else if (3 == state) { // Blink
    if (!(blink_mask & mask)) {
      blink_powersave = (blink_powersave & (0xFF ^ mask)) | (power & mask);  // Save state
      blink_power = (power >> (device -1))&1;  // Prep to Toggle
    }
    blink_timer = 1;
    blink_counter = ((!sysCfg.blinkcount) ? 64000 : (sysCfg.blinkcount *2)) +1;
    blink_mask |= mask;  // Set device mask
    mqtt_publishPowerBlinkState(device);
    return;
  }
  else if (4 == state) { // No Blink
    byte flag = (blink_mask & mask);
    blink_mask &= (0xFF ^ mask);  // Clear device mask
    mqtt_publishPowerBlinkState(device);
    if (flag) {
      do_cmnd_power(device, (blink_powersave >> (device -1))&1);  // Restore state
    }
    return;
  }
  if (publishPower) {
    mqtt_publishPowerState(device);
  }
}

ionciubotaru, 2017-09-03:
Another solution:
`Backlog power1 0; delay 20;power2 1`
and reverse:
`Backlog power2 0; delay 20;power1 1`
Also use PulseTime to prevent one relay remaining ON for long time

lobocobra, 2017-09-03:
Good point, but then I would do it by MQTT directly....

But I wanted an idiot proof solution, where I can push the buttons on the device and on the web-site. Because everything that can go wrong, will go wrong :-)
=> what would be even more cool than my dirty hack, is to have the possibilities to run scripts like easyesp 2.0. Then you could use the backlog command as part of the script.

stefanbode, 2017-09-04:
Your enhancement makes absolute sense because not only shutter requires this behaviour. My irrigation pumps also need 2 relays to open and close and should never be ON both. I solved this with the pulsetimer, but as you already mentioned this is just a workaround. I do not need a configuration what relays belong to the group. This can be hardwired. E.g. Relay 1 and 3 one group and 2 and 4 the other. In the GPIO definition, you just add two new definitions RelaysGroup1 and RelaysGroup2 and assign them to both GPIO. The rest should be easy to implement. In this case, you don't break the original functions. I will check if I can implement this in my fork of TASMOTA.

lobocobra, 2017-09-10:
This would be even better! I will check your fork, as currently I need to apply the changes to each upgrade manually :)

I like your Idea with 2 new GPIO definitions. Unfortunately I am not yet enough in the Tasmota code nor Arduino programming, to test your proposal.

alakdae, 2017-12-23:
@lobocobra Could you help me out with a solution for this case? I think it will be similar to what you did.
https://github.com/arendst/Sonoff-Tasmota/issues/1423

alakdae, 2017-12-23:
Never mind, did it. Solution is dirty, but working great. If anyone is interested let me know.

diasarmando, 2018-02-22:
Hi,

The latest versions has a quite different code on the sonoff file.
I tryed to change it following the logic of your code but I cant put it working. After the firmware update, the behaviour is the same as without the patch.

My modifed methods:

`void SetDevicePower(power_t rpower)
{
  uint8_t state;

  if (POWER_ALL_ALWAYS_ON == Settings.poweronstate) {  // All on and stay on
    power = (1 << devices_present) -1;
    rpower = power;
  }

#ifdef Sonoff4CH_Interlock_SPLIT 0 // split your channels into 2 groups, each group allows one active channel only
// ---- changes by lobocobra ----
if (Settings.flag.interlock) { // Allow only one or no relay set
uint8_t mask = 0x01;
uint8_t count = 0;
byte result1 =0;
byte result2 =0;
for (byte i = 0; i < devices_present; i++) {
if (rpower & mask) {
if (i <2) { result1++;}//increment if low part is ON
if (i >1) { result2++;}//increment if high part is ON
}
mask <<= 1;
}
if ((result1) >1 && (result2 >1)) {power = 0; rpower = 0;}
if ((result1) >1 && (result2 <2)) {power = power & 0x0C; rpower = power;} // 1/2 are both on and 3/4 max one is on
if ((result1) <2 && (result2 >1)) {power = power & 0x03; rpower = power;} // 1/2 max one is on and 3/4 both are on
}
#else // use original code, only one channel allowed

  if (Settings.flag.interlock) {     // Allow only one or no relay set
    power_t mask = 1;
    uint8_t count = 0;
    for (byte i = 0; i < devices_present; i++) {
      if (rpower & mask) {
        count++;
      }
      mask <<= 1;
    }
    if (count > 1) {
      power = 0;
      rpower = 0;
    }
  }
#endif
// --- end of modification by lobocobra
  XdrvSetPower(rpower);

  if ((SONOFF_DUAL == Settings.module) || (CH4 == Settings.module)) {
    Serial.write(0xA0);
    Serial.write(0x04);
    Serial.write(rpower &0xFF);
    Serial.write(0xA1);
    Serial.write('\n');
    Serial.flush();
  }
  else if (EXS_RELAY == Settings.module) {
    SetLatchingRelay(rpower, 1);
  }
  else {
    for (byte i = 0; i < devices_present; i++) {
      state = rpower &1;
      if ((i < MAX_RELAYS) && (pin[GPIO_REL1 +i] < 99)) {
        digitalWrite(pin[GPIO_REL1 +i], bitRead(rel_inverted, i) ? !state : state);
      }
      rpower >>= 1;
    }
  }
}`

`void ExecuteCommandPower(byte device, byte state)
{
// device  = Relay number 1 and up
// state 0 = Relay Off
// state 1 = Relay On (turn off after Settings.pulse_timer * 100 mSec if enabled)
// state 2 = Toggle relay
// state 3 = Blink relay
// state 4 = Stop blinking relay
// state 6 = Relay Off and no publishPowerState
// state 7 = Relay On and no publishPowerState
// state 9 = Show power state

  uint8_t publish_power = 1;
  if ((POWER_OFF_NO_STATE == state) || (POWER_ON_NO_STATE == state)) {
    state &= 1;
    publish_power = 0;
  }
  if ((device < 1) || (device > devices_present)) {
    device = 1;
  }
  if (device <= MAX_PULSETIMERS) {
    pulse_timer[(device -1)] = 0;
  }
  power_t mask = 1 << (device -1);
  if (state <= POWER_TOGGLE) {
    if ((blink_mask & mask)) {
      blink_mask &= (POWER_MASK ^ mask);  // Clear device mask
      MqttPublishPowerBlinkState(device);
    }
	
	
#ifdef Sonoff4CH_Interlock_SPLIT 0 // split your channels into 2 groups, each group allows one active channel only
// ---- modifications by lobocobra ----

// find out if channel 1/2 or 3/4 are to be changed
if (device < 3 ) { // channel 1/2 are now changed
// we are on 1/2
if (Settings.flag.interlock && !interlock_mutex) { // Clear all but masked relay
interlock_mutex = 1;
for (byte i = 0; i < 2; i++) {
byte imask = 0x01 << i;
if ((power & imask) && (mask != imask)) {
ExecuteCommandPower(i +1, 0);
}
}
interlock_mutex = 0;
}
} else {
// channel 3/4 are changed
if (Settings.flag.interlock && !interlock_mutex) { // Clear all but masked relay
interlock_mutex = 1;
for (byte i = 2; i < devices_present; i++) {
byte imask = 0x01 << i;
if ((power & imask) && (mask != imask)) {
ExecuteCommandPower(i +1, 0);
}
}
interlock_mutex = 0;
}
}
#else // original code where only 1 active channel is allowed
    if (Settings.flag.interlock && !interlock_mutex) {  // Clear all but masked relay
      interlock_mutex = 1;
      for (byte i = 0; i < devices_present; i++) {
        power_t imask = 1 << i;
        if ((power & imask) && (mask != imask)) {
          ExecuteCommandPower(i +1, POWER_OFF);
        }
      }
      interlock_mutex = 0;
    }	
#endif //end of modification of lobocobra

    switch (state) {
    case POWER_OFF: {
      power &= (POWER_MASK ^ mask);
      break; }
    case POWER_ON:
      power |= mask;
      break;
    case POWER_TOGGLE:
      power ^= mask;
    }
    SetDevicePower(power);
#ifdef USE_DOMOTICZ
    DomoticzUpdatePowerState(device);
#endif  // USE_DOMOTICZ
    if (device <= MAX_PULSETIMERS) {
//      pulse_timer[(device -1)] = (power & mask) ? Settings.pulse_timer[(device -1)] : 0;
      pulse_timer[(device -1)] = (((POWER_ALL_OFF_PULSETIME_ON == Settings.poweronstate) ? ~power : power) & mask) ? Settings.pulse_timer[(device -1)] : 0;
    }
  }
  else if (POWER_BLINK == state) {
    if (!(blink_mask & mask)) {
      blink_powersave = (blink_powersave & (POWER_MASK ^ mask)) | (power & mask);  // Save state
      blink_power = (power >> (device -1))&1;  // Prep to Toggle
    }
    blink_timer = 1;
    blink_counter = ((!Settings.blinkcount) ? 64000 : (Settings.blinkcount *2)) +1;
    blink_mask |= mask;  // Set device mask
    MqttPublishPowerBlinkState(device);
    return;
  }
  else if (POWER_BLINK_STOP == state) {
    byte flag = (blink_mask & mask);
    blink_mask &= (POWER_MASK ^ mask);  // Clear device mask
    MqttPublishPowerBlinkState(device);
    if (flag) {
      ExecuteCommandPower(device, (blink_powersave >> (device -1))&1);  // Restore state
    }
    return;
  }
  if (publish_power) {
    MqttPublishPowerState(device);
  }
}`

Could someone help to understand what happens?

Tks

diasarmando, 2018-02-25:
Hi,
The latest versions has a quite different code on the sonoff file.
Ichanged it following the logic of your code but I cant put it working. After the firmware update I activate the Interlock by command and then the sonoff stay unresponsive and boot looping. The only way is flashing again the code.

My modifed methods:

void SetDevicePower(power_t rpower)
{
  uint8_t state;

  if (POWER_ALL_ALWAYS_ON == Settings.poweronstate) {  // All on and stay on
    power = (1 << devices_present) -1;
    rpower = power;
  }

#ifdef Sonoff4CH_Interlock_SPLIT 0 // split your channels into 2 groups, each group allows one active channel only
// ---- changes by lobocobra ----
if (Settings.flag.interlock) { // Allow only one or no relay set
uint8_t mask = 0x01;
uint8_t count = 0;
byte result1 =0;
byte result2 =0;
for (byte i = 0; i < devices_present; i++) {
if (rpower & mask) {
if (i <2) { result1++;}//increment if low part is ON
if (i >1) { result2++;}//increment if high part is ON
}
mask <<= 1;
}
if ((result1) >1 && (result2 >1)) {power = 0; rpower = 0;}
if ((result1) >1 && (result2 <2)) {power = power & 0x0C; rpower = power;} // 1/2 are both on and 3/4 max one is on
if ((result1) <2 && (result2 >1)) {power = power & 0x03; rpower = power;} // 1/2 max one is on and 3/4 both are on
}
#else // use original code, only one channel allowed

  if (Settings.flag.interlock) {     // Allow only one or no relay set
    power_t mask = 1;
    uint8_t count = 0;
    for (byte i = 0; i < devices_present; i++) {
      if (rpower & mask) {
        count++;
      }
      mask <<= 1;
    }
    if (count > 1) {
      power = 0;
      rpower = 0;
    }
  }
#endif
// --- end of modification by lobocobra
  XdrvSetPower(rpower);

  if ((SONOFF_DUAL == Settings.module) || (CH4 == Settings.module)) {
    Serial.write(0xA0);
    Serial.write(0x04);
    Serial.write(rpower &0xFF);
    Serial.write(0xA1);
    Serial.write('\n');
    Serial.flush();
  }
  else if (EXS_RELAY == Settings.module) {
    SetLatchingRelay(rpower, 1);
  }
  else {
    for (byte i = 0; i < devices_present; i++) {
      state = rpower &1;
      if ((i < MAX_RELAYS) && (pin[GPIO_REL1 +i] < 99)) {
        digitalWrite(pin[GPIO_REL1 +i], bitRead(rel_inverted, i) ? !state : state);
      }
      rpower >>= 1;
    }
  }
}

void ExecuteCommandPower(byte device, byte state)
{
// device  = Relay number 1 and up
// state 0 = Relay Off
// state 1 = Relay On (turn off after Settings.pulse_timer * 100 mSec if enabled)
// state 2 = Toggle relay
// state 3 = Blink relay
// state 4 = Stop blinking relay
// state 6 = Relay Off and no publishPowerState
// state 7 = Relay On and no publishPowerState
// state 9 = Show power state

  uint8_t publish_power = 1;
  if ((POWER_OFF_NO_STATE == state) || (POWER_ON_NO_STATE == state)) {
    state &= 1;
    publish_power = 0;
  }
  if ((device < 1) || (device > devices_present)) {
    device = 1;
  }
  if (device <= MAX_PULSETIMERS) {
    pulse_timer[(device -1)] = 0;
  }
  power_t mask = 1 << (device -1);
  if (state <= POWER_TOGGLE) {
    if ((blink_mask & mask)) {
      blink_mask &= (POWER_MASK ^ mask);  // Clear device mask
      MqttPublishPowerBlinkState(device);
    }
	
	
#ifdef Sonoff4CH_Interlock_SPLIT 0 // split your channels into 2 groups, each group allows one active channel only
// ---- modifications by lobocobra ----

// find out if channel 1/2 or 3/4 are to be changed
if (device < 3 ) { // channel 1/2 are now changed
// we are on 1/2
if (Settings.flag.interlock && !interlock_mutex) { // Clear all but masked relay
interlock_mutex = 1;
for (byte i = 0; i < 2; i++) {
byte imask = 0x01 << i;
if ((power & imask) && (mask != imask)) {
ExecuteCommandPower(i +1, 0);
}
}
interlock_mutex = 0;
}
} else {
// channel 3/4 are changed
if (Settings.flag.interlock && !interlock_mutex) { // Clear all but masked relay
interlock_mutex = 1;
for (byte i = 2; i < devices_present; i++) {
byte imask = 0x01 << i;
if ((power & imask) && (mask != imask)) {
ExecuteCommandPower(i +1, 0);
}
}
interlock_mutex = 0;
}
}
#else // original code where only 1 active channel is allowed
    if (Settings.flag.interlock && !interlock_mutex) {  // Clear all but masked relay
      interlock_mutex = 1;
      for (byte i = 0; i < devices_present; i++) {
        power_t imask = 1 << i;
        if ((power & imask) && (mask != imask)) {
          ExecuteCommandPower(i +1, POWER_OFF);
        }
      }
      interlock_mutex = 0;
    }	
#endif //end of modification of lobocobra

    switch (state) {
    case POWER_OFF: {
      power &= (POWER_MASK ^ mask);
      break; }
    case POWER_ON:
      power |= mask;
      break;
    case POWER_TOGGLE:
      power ^= mask;
    }
    SetDevicePower(power);
#ifdef USE_DOMOTICZ
    DomoticzUpdatePowerState(device);
#endif  // USE_DOMOTICZ
    if (device <= MAX_PULSETIMERS) {
//      pulse_timer[(device -1)] = (power & mask) ? Settings.pulse_timer[(device -1)] : 0;
      pulse_timer[(device -1)] = (((POWER_ALL_OFF_PULSETIME_ON == Settings.poweronstate) ? ~power : power) & mask) ? Settings.pulse_timer[(device -1)] : 0;
    }
  }
  else if (POWER_BLINK == state) {
    if (!(blink_mask & mask)) {
      blink_powersave = (blink_powersave & (POWER_MASK ^ mask)) | (power & mask);  // Save state
      blink_power = (power >> (device -1))&1;  // Prep to Toggle
    }
    blink_timer = 1;
    blink_counter = ((!Settings.blinkcount) ? 64000 : (Settings.blinkcount *2)) +1;
    blink_mask |= mask;  // Set device mask
    MqttPublishPowerBlinkState(device);
    return;
  }
  else if (POWER_BLINK_STOP == state) {
    byte flag = (blink_mask & mask);
    blink_mask &= (POWER_MASK ^ mask);  // Clear device mask
    MqttPublishPowerBlinkState(device);
    if (flag) {
      ExecuteCommandPower(device, (blink_powersave >> (device -1))&1);  // Restore state
    }
    return;
  }
  if (publish_power) {
    MqttPublishPowerState(device);
  }
}

Could someone help to understand what happens?

Tks

B1G1, 2018-03-25:
very good, I'm using a sonoff 4ch to control 2 roller shutters, and i need a interlocking mode working with group of two channels.
@diasarmando can you explain better what I need to replace in the original code?

diasarmando, 2018-03-26:
you need to edit the SONOFF file and replace the SetDevicePower(power_t rpower) and ExecuteCommandPower(byte device, byte state)  code for the suggested above methods.

B1G1, 2018-03-26:
No need to ad this?

In user_config_overide.h add the following:
#define Sonoff4CH_Interlock_SPLIT 0

diasarmando, 2018-03-26:
Yes its needed.

B1G1, 2018-03-27:
Ok, can i update it from the administration page or i have to flash directly throug the serial?

diasarmando, 2018-03-27:
You need to build and upload (serial or OTA)  the firmware with the code modifications.

B1G1, 2018-04-04:
ho when i try to upload the compiled bin I get this error 
Upload Failed

Upload buffer miscompare

I read in other threads that the problem is a bad configuration of flash option, either 512 flash or no or to much spiffs, but in my Arduino ide is 1M (no SPIFFS) and it should be correct.

What's the problem?

lobocobra, 2018-04-18:
Did you try this?
https://github.com/arendst/Sonoff-Tasmota/issues/1539

Assuming that you double checked your settings with the wiki.... load a minimal version first, and then your patched one...

B1G1, 2018-04-24:
Yes! It worked!
thank you very much.

MadDoct, 2018-05-25:
Can this modification be used with a generic module (wemos d1 mini) with 4 relays? In the code it seems to me that it can only be used in the sonoff dual and 4ch... What modifications would be needed?

localhost61, 2018-06-13:
@lobocobra 
I already use 8 Sonoff Dual R2 to control my roller shutters and they work great in Tasmota v6.00a. I also bought two CH4 R2 to control the remaining 4 roller shutters but I'm missing the GPIO pins of the Dual R2 to connect my wall switches for a good WAF ;-) .  
- Did you connect manual switches on your CH4 (not Pro) to operate the shutters?
- Did you convert the 4 buttons into switches for that purpose?
I want to use your mod to be able to operate both shutters together, but still protecting the motors with the interlock feature.

Regarding your [problems for flashing](https://github.com/lobocobra/sonoff-brenner/blob/master/Sonoff%20Howto%20RRS.txt], I experimented that with CH4 R2 and Dual R2 (both are ESP8255) you don't have to enter Programming Mode like other ESP8266 but keep GPIO0 at Gnd during flashing.

lobocobra, 2018-06-22:
As I had some request to update the function see here the working update on version 6.00a.
If you want to use it on CH4-Pro then make following settings on the board:
=> S6 on position 1
=> K5 all switch position ON
=> K6 all position OFF

**user_conrig.h** 
```
// start modification lobocobra
#define Sonoff4CH_Interlock_SPLIT 0
// end modification lobocobra 
```

lobocobra, 2018-06-22:
@localhost61

about your questions....
   

>  Did you connect manual switches on your CH4 (not Pro) to operate the shutters?
=> I mounted two CH4-pro in an electric box to the wall in my garage, where I could press the buttons
=> I bought four Intertechno Funk-Wandsender YWT-8500 and programmed the CH4-pro 433mhz to it
=> I have tablets on the wall and openhab-panel running to command the shutter
=> I have rules in openhab that close / open the shutter depending on rain, wind and sun elevation

>  Did you convert the 4 buttons into switches for that purpose?
=> I wanted a clean installation and manly use the Intertechno switches and openhab to operate the shutters.

>  I want to use your mod to be able to operate both shutters together, but still protecting the motors with the interlock feature.
=> You probably want to set PULSETIME1-4 to someting like 15 seconds, so if anything goes wrong, your motors stop after 15 seconds.

stefanbode, 2018-06-22:
I make a proposal how to support the shutter usecase, without changing adding to much code to it. The functionality is already added to my branch. See stefanbode mod.

Idea:
Added SetOption31 to enhance the interlock mode (SetOption14 must also be ON) to a paired version. This pairs ALL relays in groups of two. After using the option only relay 1,3,5,7... can be actively switched. Relay 2,4,6,... will be switched as a slave with the opposite of the corresponding relay.

Main change below in the interlock. PulseTime still works and set the corresponding relay after period

```
  //STB mode
  if (Settings.flag.interlock) {
    if (Settings.flag.paired_interlock) {
      for (byte i = 0; i < devices_present; i=i+2) {
        bitWrite(rpower, i+1, !bitRead(rpower, i));
      }
    } else {
      power_t mask = 1;
      uint8_t count = 0;
      for (byte i = 0; i < devices_present; i++) {
        if (rpower & mask) count++;
        mask <<= 1;
      }
      if (count > 1) {
        power = 0;
        rpower = 0;
      }
    }
  }
//end
```

localhost61, 2018-06-25:
@lobocobra 
Many thanks for you update, I'll give it a try for sure !
@stefanbode 
In my mind, shutters are 3-state: UP/DN/OFF
If I understand properly, your mod only allow UP/DN, isn't it?
OFF may be achieved with Pulstime, but it wouldn't allow open at 20%, for instance.

stefanbode, 2018-06-26:
Yes, I was already thinking about this. The problem is not doing it, the problem is doing it in a smartbway without adding to much code. Therefore using something that already exist is a good idea. A trhree state switch is nothing already be there. Just preventing two relays on is also something that you currently can simulate quite nice with the pulsetime, as long as you do not decide to switch during this period like my wife did. As far as I cannsee the problem of the interlock is that you need to remember the old state to decide to switch off wich relay if inthronisiert request both are on. If we solve this on the UI and the API that’s fine too.

stefanbode, 2018-06-26:
Let me propose a simple solution. All,working in the group. If both relay OFF, all fine do nothing. If relay 1 or 2 ON also finen do nothing. If relay 1 and 2 are ON always switch OFF the second relay. Therefore you prevent mess but you cannot just move in the opposite direction by switching relay 2 to ON. You ,ist switch relay 1 off upfront manually or by a pulsetimer. If ok i can give it a try in my code and share results

stefanbode, 2018-06-26:
ok, done and checked in into my branch. Changes are small and smart. here is how I did it. Honestly, easy way is just use my branch :-).

 In sonoff.ino
Change Function ExecuteCommandPower snipped below. 
```
    if (Settings.flag.interlock && !interlock_mutex ) {  // Clear all but masked relay
      interlock_mutex = 1;
    //stb mod
      if (Settings.flag.paired_interlock) {
        byte  temp = 2* !(device%2);
        power_t imask = 1 << (device- temp);
        //snprintf_P(log_data, sizeof(log_data), PSTR("SRC0: device %d, state: %d, source %d, devicebin: %d, power: %ld, mask: %ld, imask %ld"),device,state,source, device%2 ,power, mask, imask);
        //AddLog(LOG_LEVEL_DEBUG);
        if (power & imask) ExecuteCommandPower(device +1-temp , POWER_OFF, SRC_IGNORE);
      } else {
        for (byte i = 0; i < devices_present; i++) {
          power_t imask = 1 << i;
          if ((power & imask) && (mask != imask)) ExecuteCommandPower(i +1, POWER_OFF, SRC_IGNORE);
        }
      }
      //end
      interlock_mutex = 0;
    }
```
Additionally the wiping of all relays must be more conditional: change SetDevicePower
replace condition:
`  if (Settings.flag.interlock) {`
with
`  if (Settings.flag.interlock && !Settings.flag.paired_interlock) {`

lobocobra, 2018-06-26:
Hey stefanbode

I like your idea to channge as little code as possible, but your solution is almost obsolet if you use an CH4pro as each Swich can be connected with 2 cables, where one is always on.
=> So using a CH4pro you could attach 4 shutter, but only select if they go UP or DOWN and you can not stop then.

This is why we would need to alter your function, so that it allows to stop a shutter and not to have only the option to have it up/down but up/stop/down.

markvenbrux, 2018-06-26:
Hi,
CH4pro is nice for this: when you use two relays for one shutter, you can
power the second relay from the Normally Closed output of the first relay.
Then the shutter motor is physically protected, and you have the
up/stop/down functionality.
Cheers,
Mark

stefanbode, 2018-06-26:
Yes also a good idea, but then you have to take care of direction with one relay and motor on/off with the other. Maybe more overhead, if controlling it by MQTT or in something like openhab. Anyhow, I like the idea because it is failsafe by nature.

stefanbode, 2018-06-26:
@lobocobra with the current implementation a stop is also working. see example in my wiki. Only both relays ON in an invalid option that causes the other relay to switch. instead of this it behaves in the current implementation like a normal relay

lobocobra, 2018-06-26:
@stefanbode 
Thanks for the hint. I did not realise that your solution accepts OFF/OFF.
=> But does this not mean that ON/ON can be activated for a fraction of a second as well? 
=> My shutters would then go directly into programming mode and the mess is starting.

My plans go even further. I want SONOFF to handle the position of the shutter. So I send the mqtt command 30% and SONOFF brings it to that position. This means that I will need to add a few variables and to save the current position. And of course I will need to add some MQTT commands. 
=> once done we can talk about creating a module, which could increase the probablilty that arendst consider to add it to the master.

stefanbode, 2018-06-26:
In the code there is definitely a secure function to ensure one relay get of BEFORE the other gets on. Now we have to talk about physics and how long a mechanical relays takes to switch from one state to another. If both times are equal, no problem, but if switching off takes 0.3sek and swtiching on only 0.1, uhhhhh, you're in trouble because the code is fast. If you want ensure you can add a delay(200) for example after the Switch OFF command.

```
if (power & imask) {
   ExecuteCommandPower(device +1-temp , POWER_OFF, SRC_IGNORE);
   delay(200);
}
```

B1G1, 2018-06-26:
Hi all, 
i'm interested about the possibility to handle the position of the shutter by the sonoff.
On the home assistant forum kabongsteve has done an extension for that.
https://community.home-assistant.io/t/sonoff-tasmota-extension-for-covers/44694

May be you could start from that.

stefanbode, 2018-06-26:
Regarding your idea with the 30% and then let it run is great, but really nasty to implement. You have to work with pulsetimer to exactly controll motor run time. From outside switching is not accurate enough. Now the use case from my wife: shutter 80% just in the middle during operation. "hm, to much 40%" Expectation immediate stop and follow the new instructions like I do :-). This is really hard to do inside the firmware because you need to cancel the pulsetimer and recalculate the current runtime=position of the shutter before calculatiing the duration and direction.

stefanbode, 2018-07-02:
ok, done. After some hours coding, this is now reliably working in my mod. All calculation is done in the firmware and you can change the target position at any time, even it the blind is still operating. I support different times for OPEN and CLOSE but have currently no startuptime that might reflect the delay until the blind starts to move. Let me know if this works for you in the current configuration.

lobocobra, 2018-07-03:
oooops ... well I finished mine as well :-). But congrats as well to @stefanbode. 

**Currently I support following commands, all can either set or request the value from the sonoff:**
 * - ShutterMinSec:   minimum position of the shutter in seconds ( my stopper does not work...)
 * - ShutterMaxSec   maximum position of the shutter in seconds ( how long I run the motor)
 * - shutterPosSec    what is the current position of the shutter in seconds
 * - shutterGoPercent  order to let move the shutter to the right position
 * - shutterStartDelay  what delay does the motor has to start (important if you stop,go,stop,go)
Those commands can be sent either per mqtt or in the console for Shutter 1/2 and 3/4 (pair =down).

Of course you can change direction anytime during operation and it keeps as well track of it. :-)
=> I did not yet foresee any different times for up/down, but you are right this is probably needed.
=> I am unsure if STOP needs also a delay, currently the delay is only applied to START

stefanbode, 2018-07-03:
Please share your code and let’s work on a common solution. The reason for different open and close time is mainly for blinds. If the motor has to open there is load on the system that slows down the opening process. This is up to 30% depending on the weight of the blinds. If you overengineer it you should change the factor linear over the open position. I just leave it constant. OpenHAB is working, now I need some help for homeassistant and domotricz for easy integration. Also the report of the current position must be integrated into the teleperiod state report. 
domotricz.

B1G1, 2018-07-03:
Great work guys!
have you planned the possibility to use momentary switches?
in my case i need a pulse to start the rolling shutter and a pulse if i want to stop it before it’s fully open or fully closed

stefanbode, 2018-07-03:
This will also work without any modification. Just define a short pulsetimer on the both relays should fulfill your requirements

B1G1, 2018-07-03:
@stefanbode ok, and can i manage an initial delay only when i open the roller shutter from fully closed?
You can watch this video to understand what I mean https://youtu.be/Ifp-bqG9y54

lobocobra, 2018-07-10:
Just as an update....
I have now a working version, where I added additional commands to MQTT like:
 * - ShutterMinDSec1/2 (enter DeciSec = Sec*10)
 * - ShutterMaxSec1/2
 * - shutterPosSec1/2
 * - shutterGoPercent1/2
 * - shutStartDelayDSec1/2 (enter DeciSec = Sec*10)
 * - shutLagUpwardsDSec1/2 (enter DeciSec = Sec*10)

You can operate the shutter by: buttons, mqtt, the web-interface. And it stops automatically, when you are MIN or MAX.  
=> Why I did it not release yet is because I have minor bugs in, which have the effect, that the shutter is not going fully up and it stays 5% out, while the sonoff thinks it is fully up. 

I am gradually improving this and consider to move from using my own TimerVariable to UTC and not to use anymore PulseTimer, which causes the problem.  The current version would only usable anyway for systems wherre you do NOT need a pulse to start/stop but a continous ON of a relay.
=> So when I use UTC and not anylonger PulseTime that limitation would be gone as well.

lobocobra, 2018-07-14:
Here we are, for those who are interested: https://github.com/lobocobra/sonoff-shutter
Following commands are supported....
 *  ShutterMinDSec1/2 (enter DeciSec = Sec*10) : minimal position of the shutter
 *  ShutterMaxDSec1/2 (enter DeciSec = Sec*10) : maximum position of the shutter
 *  shutterPosDSec1/2 (enter DeciSec = Sec*10) : current position in DSEC (corrected)
 *  shutterGoPercent1/2 (enter DeciSec = Sec*10) : Move shutter to x%
 *  shutStartDelayDSec1/2 (enter DeciSec = Sec*10) : time the motor needs to start the shutter in Dsec
 *  shutLagUpwardsDSec1/2 (enter DeciSec = Sec*10) : difference in Dsec the shutter needs longer UP

Commands can be entered on MQTT, Web, button. The sonoff will figure out all the rest. 
This is only an add-on and I will probably not keep it up to date with future releases (only if it is needed for me). But I have commented what I did in the code. Search for LOBOCOBRA and you will understand my code.

stefanbode, 2018-07-14:
Where did you put the code. I'm missing something like xdrv_99_... ino. All additional capabilities should be separated into files to ensure compatibility and the option to add/remove the code from compiling. I agree that some changes are required inside the master code, but the changes here should be as less as possible. Can you explain why you have SchutterMin and SchutterMax and shutterLagUp? Normally Min should always be 0. == fully open. @ascillato with the current coding a pull request is difficult.

ascillato, 2018-07-14:
@stefanbode Yes, you are totally right.

@lobocobra Can you isolate your modifications so a pull request can be made?

stefanbode, 2018-07-14:
I see that you work with pulsetimers. This was my first attempt, too, but it has many disadvantages. The most important one, is that you cannot make any change during operation. Therefore I decide in my implementation not to use it. Additionally, you will get into trouble, if you want to integrate two switches to drive the shutter manually. One additional problem that you will find sooner or later is, that there are shutter tapes that work with pulses instead of continuous ON /OFF. This will also make it hard with the pulsetime. All of this persuades me to track position on my own. [https://github.com/stefanbode/Sonoff-Tasmota/blob/master/sonoff/xdrv_97_shutter.ino](https://github.com/stefanbode/Sonoff-Tasmota/blob/master/sonoff/xdrv_97_shutter.ino)

ascillato, 2018-07-14:
Ah, great. You have made as a driver! So, could you make a pull request with your changes?

stefanbode, 2018-07-14:
Last changes comitted yesterday. Lets wait for some user feedback and fix bugs, if there are still some in. Later on a pull request is a good idea. Normally I wait 2-4 weeks to ensure the code is stable and optimized; at least that @arendst take a look at it :-).

lobocobra, 2018-07-14:
@all 
Thanks for your input, I will work on it, bit by bit.

@stefanbode 
Regarding pulsetime (please see latest update I made 10min ago).
I solved the problem differently and you can start/stop in the middle of the opertion, while it uses pulsetime. I might change that in future but did not see any need yet. Also additional switches should not be be an issue (not yet testet), but you can stop the shutter anytime and the code should handle the problems.
=> Where it does really not work is with shutter that only request an impulse for start/stop. But if this is really requested, I can change the code quickly as it needs pulsetime only for the GOpercent command.

Minimum Value
My shutter has the problem that it goes too much up. So I added the possibilty of a minimum level. The shutter stops at the minimum and the maximum, whatever pulsetime is set. 
=>

lobocobra, 2018-07-14:
Explanation of:
* ShutterMin
My shutter does have an issue and is going too much up. So I added the possibility to stay above 0.

* ShutterMax
This is the time needed to go from 0 (not Min) to Max. It will always stop at this position.
* shutterLagUp 
This is the time difference between going DOWN and UP. So if your shutter needs 3.5 seconds longer to go up, set this value to 35.

lobocobra, 2018-07-16:
I moved now the code to a driver page, in order to update it more easily. I could also reduce a bit the code and it takes now complied 455k. 
=> My plans are to extend it to calculate also how much water my garden is using (electric pump). 
=> In addition I will use it as basis for my heating counter (define how much oil is left).

localhost61, 2018-07-17:
@lobocobra 
I tried but couldn't compile your project.
File xdrv_91.ino is missing so it fails to find ExecutePowerUpdateShutterPos().
platformio.ini was missing too

lobocobra, 2018-07-17:
Thanks for the feedback. I re-uploaded the project.
=> can you check it again?

localhost61, 2018-07-17:
Yes, I could compile.
The only weird thing is that platform.io couldn't locate _Adafruit_Sensor.h_, so I turn around the issue by putting a copy from _Adafruit_Sensor-1.0.2.02_ into _Adafruit_BME680-1.0.5_

Do you think I'll get an issue if I test on a Dual R2? My CH4 R2 devices are still not installed.

localhost61, 2018-07-17:
It's cosmetic but in `user_config.h`, you don't need to assign a value to define `Sonoff4CH_Interlock_SPLIT` because in `sonoff.ino` the `#ifdef` directive test if `Sonoff4CH_Interlock_SPLIT` is defined, not it's value.

So you should remove the 0 value assignment in both file to remove both warning at 
```
- sonoff.ino:323:35 & 
- sonoff.ino:1270:34: warning: extra tokens at end of #ifdef directive
```

lobocobra, 2018-07-17:
Thanks for the input, you are right, with assigning the value, its an obsolete piece of code I made a year ago.
=> As I am currently removing that part completely and will replace it by  SETOPTION81 ,  this code is already gone in the dev section. 

For testing I see no big issue as long you do not send commands for the Shutter2. But I probably need to fix that anyhow in the next version. Of course it can make sense to use as well an DUAL Sonoff.

lobocobra, 2018-07-30:
updated to latest arendst version

speedymk1, 2018-08-06:
Hi,
I will also need a shutter solution for the sonoff 4CH and found this.
If I understood your comments, you have already done a firmware.
But I need it to switch via MQTT, WEB,... and also via Buttons.
This means:
Btn1 -> shutter 1 up (as long as you press the butten it goes up)
Btn2 -> shutter 1 down (as long as you press the butten it goes down)
Btn 3 -> like Btn1, but for shutter 2
Btn 4 -> like Btn2, but for shutter 2
Additional if the shutter goes up (Btn1 was pressed) and you press Btn2 (shutter 1 down) short,
the shutter only stop at the actual position.

Could I do this with your firmware?
If yes, but also if no, is there a bin file for OTA update?

Thanks a lot in advance
regards
Speedy

stefanbode, 2018-08-07:
Hi speedymk1. A side development (branch) can help you. https://github.com/stefanbode/Sonoff-Tasmota . If you need a BIN file for the 4CH let me know. This supports up to 4 shutters, if you have the hardware :-)

lobocobra, 2018-08-07:
@speedymk1 

Yes, my solution offers the requested functuonality you requested. You will find it here:
 [https://github.com/lobocobra/sonoff-shutter]

I send you a bin if you wish.

[25 more replies]