arendst/Tasmota #829: So off Basic. Not flashing.

eggfat, 2017-09-04:
Hi there, I'm trying to flash some basics (dated 2017 6 20.) using Platformio. lve tried flashing many times but I'm getting nowhere. Is this a known issue? Anyone else having the same problem? Any help would be much appreciated.

davidelang, 2017-09-04:
you provide no details, so we can't begin to guess what's wrong

yes, lots of people have reported problems flashing, but lots of other people 
have no problems.

are you using DOUT or DIO (you need to use DOUT)

are you sure you are getting the device into flashing mode? (connecting GPIO0 
to ground before applying power)?

are you sure you aren't using a 5v serial interface?

eggfat, 2017-09-04:
Hi there, 
thanks for the reply, yep I was being a bit vague.
Im using Atom with PlatformIO.
3.3v Serial interface (which I have flashed 3 other sonoff's with)
DOUT is in the script? or is there an option to select in Atom PlatformIO?

Thanks for any help.
Here is the error Im getting.

73.7 s
platformio run --target upload
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266WiFi/ESP8266WiFiMulti.o
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266WiFi/ESP8266WiFiSTA.o
Archiving .pioenvs/sonoff-ds18x20/lib/libOneWire.a
Indexing .pioenvs/sonoff-ds18x20/lib/libOneWire.a
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266WiFi/ESP8266WiFiScan.o
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266WiFi/WiFiClient.o
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266WiFi/WiFiClientSecure.o

Archiving .pioenvs/sonoff-ds18x20/lib/libIRremoteESP8266.a
Indexing .pioenvs/sonoff-ds18x20/lib/libIRremoteESP8266.a
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266WiFi/WiFiServer.o
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266WiFi/WiFiUdp.o
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266HTTPClient/ESP8266HTTPClient.o
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266httpUpdate/ESP8266httpUpdate.o
Compiling .pioenvs/sonoff-ds18x20/lib/Ticker/Ticker.o
Compiling .pioenvs/sonoff-ds18x20/lib/DNSServer/DNSServer.o
Archiving .pioenvs/sonoff-ds18x20/lib/libTicker.a
Indexing .pioenvs/sonoff-ds18x20/lib/libTicker.a
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266mDNS/ESP8266mDNS.o
Archiving .pioenvs/sonoff-ds18x20/lib/libESP8266WiFi.a
Archiving .pioenvs/sonoff-ds18x20/lib/libDNSServer.a
Archiving .pioenvs/sonoff-ds18x20/lib/libESP8266httpUpdate.a
Compiling .pioenvs/sonoff-ds18x20/lib/Wire/Wire.o
Indexing .pioenvs/sonoff-ds18x20/lib/libESP8266WiFi.a
Indexing .pioenvs/sonoff-ds18x20/lib/libESP8266httpUpdate.a
Indexing .pioenvs/sonoff-ds18x20/lib/libDNSServer.a
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266WebServer/ESP8266WebServer.o
Compiling .pioenvs/sonoff-ds18x20/lib/ESP8266WebServer/Parsing.o
Archiving .pioenvs/sonoff-ds18x20/lib/libArduinoJson.a
Indexing .pioenvs/sonoff-ds18x20/lib/libArduinoJson.a
Archiving .pioenvs/sonoff-ds18x20/lib/libESP8266HTTPClient.a
Archiving .pioenvs/sonoff-ds18x20/lib/libWire.a
Indexing .pioenvs/sonoff-ds18x20/lib/libESP8266HTTPClient.a
Indexing .pioenvs/sonoff-ds18x20/lib/libWire.a
Archiving .pioenvs/sonoff-ds18x20/lib/libESP8266mDNS.a
Indexing .pioenvs/sonoff-ds18x20/lib/libESP8266mDNS.a
Archiving .pioenvs/sonoff-ds18x20/lib/libESP8266WebServer.a
Indexing .pioenvs/sonoff-ds18x20/lib/libESP8266WebServer.a
Linking .pioenvs/sonoff-ds18x20/firmware.elf
Checking program size
text       data     bss     dec     hex filename
477957     8276   35136  521369   7f499 .pioenvs/sonoff-ds18x20/firmware.elf
Building .pioenvs/sonoff-ds18x20/firmware.bin
Looking for upload port...
Auto-detected: /dev/cu.usbserial-A600CIVX
Uploading .pioenvs/sonoff-ds18x20/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255
========================== [ERROR] Took 19.32 seconds ==========================
 
================================== [SUMMARY] ==================================
Environment sonoff              [ERROR]
Environment sonoff-NL           [ERROR]
Environment sonoff-minimal      [ERROR]
Environment sonoff-ds18x20      [ERROR]
========================== [ERROR] Took 73.12 seconds ==========================

eggfat, 2017-09-05:
Can anyone help? Ive been searching the internet but Im getting nowhere.
Ive uncommented env_default = sonoff.
Im holding down the power to get into flashing mode before powering.
Hitting PlatformIO - upload (Atom) 
Getting errors.
collect2: error: ld returned 1 exit status
*** [.pioenvs/sonoff/firmware.elf] Error 1

ianbee86, 2017-09-05:
Hey Eggfat,

I would personally start with trying to connect to the device.

1. Start your sonoff Basic with out putting in programming mode
2. Connect using putty or terminal (screen) via your Serial Device to your Sonoff Basic

More than likely what is happening is there is no connection or the device is faulty.

What happens when you just boot the device do you get the blue flashing light and can you see the Wifi Access point?

eggfat, 2017-09-05:
Hi, I can connect it to the EweLink app while its connected to the FTDI module and have full control. 
Its when I try to flash it Im getting the errors.
The light on this unit is green. Should it flash blue when in programming mode?
All the other sonoffs flashed fine its just these two new that Im having problems with.
Thanks.

davidelang, 2017-09-06:
I don't know how to set the mode with platformIO, try flashing it with the 
Arduino IDE (set to DOUT), just to make sure you have everything working, then 
you can go back to figuring out how to do it with platformio

David Lang

eggfat, 2017-09-06:
Hi,
I started from scratch. Set up Arduino with the instructions from the wiki.
And it flashed 100%
So I set up the MQTT parameters changed host and topic hit save and it died.
Tried flashing again. got this error.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
Tried the other sonoff.
Same error.

Read that it may not have enough power though the FTDI so I wired it up to see if I could get a wifi signal but wouldn't switch on.

Im really not sure. Any suggestions?

Thanks.

eggfat, 2017-09-13:
Couldn't get these two to work unfortunately. Must have been a bad batch?

Bought two more and they flashed straight away.

Thanks.

closed 2018-05-07 by stale[bot]