arendst/Tasmota #830: Fatal exception while browsing web interface

c2342, 2017-09-04:
Hi, 

I've just updated my S20 and new RF bridge to 5.6.1k but when I try to access the web interface I get the following error below:

```
{```
"Module":"Sonoff Basic", "Version":"5.6.1k", "FallbackTopic":"DVES_###", "GroupTopic":"sonoffs"}
{"WebServerMode":"Admin", "Hostname":"sonoff-0506", "IPAddress":"###"}
{"RestartReason":"Fatal exception:3 flag:2 (EXCEPTION) epc1:0x4000bef4 epc2:0x00000000 epc3:0x00000000 excvaddr:0x402674da depc
{"POWER":"ON"}
ON
{"Time":"2017-09-04T11:19:11", "Uptime":0, "Vcc":3.154, "POWER":"ON", "Wifi":{"AP":1, "SSId":"###", "RSSI":100, "APMac":"###"}}
Offline
Online
{"Module":"Sonoff Basic", "Version":"5.6.1k", "FallbackTopic":"DVES_###", "GroupTopic":"sonoffs"}
{"WebServerMode":"Admin", "Hostname":"sonoff-0506", "IPAddress":"###"}
{"RestartReason":"Fatal exception:3 flag:2 (EXCEPTION) epc1:0x4000bf64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x40267415 depc
{"POWER":"ON"}
**ON**
```
Awesome work otherwise!

Brandon2255p, 2017-09-04:
Same issue here. I have never used this firmware before so I could be doing something wrong in the user config. I also changed options in the ini file because I am using the electrodragon relay board.
```

Exception (3):
epc1=0x4000bef4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4025cae2 depc=0x00000000
 

ctx: cont
sp: 3fff14a0 end: 3fff1860 offset: 01a0
 
>>>stack>>>
3fff1640:  00000005 00000000 3fff1680 402022d1
3fff1650:  3ffefa29 00000001 3fff1680 40202322
3fff1660:  4025cae2 00000001 4025cae2 40202568
3fff1670:  00000001 00000001 4025cae2 40217dd2
3fff1680:  3fff414c 0000000f 00000009 4010053d
3fff1690:  00000010 3fff1750 00000000 4010053d
3fff16a0:  3fff18d4 000003c6 000003c6 4010020c
3fff16b0:  00000000 3ffe8d3b 00000001 3fff18d4
3fff16c0:  00000010 3fff1750 00000000 4010053d
3fff16d0:  402284e2 00000000 3fff1750 402284f9
3fff16e0:  3fff4024 3fff1750 3fff1750 3ffe8d3b
3fff16f0:  00000010 3fff1750 3fff1750 3fff18d4
3fff1700:  00000010 3fff1790 00000000 4010053d
3fff1710:  3fff18d4 00000458 00000458 3fff18d4
3fff1720:  3fff42b4 0000000f 00000003 3fff4644
3fff1730:  000005cf 000005c0 3fff1790 4020224c
3fff1740:  3fff4024 00000001 3fff1790 4020229b
3fff1750:  3fff18d4 00000418 00000418 40207914
3fff1760:  00000001 00000001 3fff2d9c 40219ed6
3fff1770:  00000000 3fffdad0 3fff2d9c 4020790a
3fff1780:  3fff2d9c 3fff2fc4 3fff2d9c 40207946
3fff1790:  00000000 00000000 00000000 40202400
3fff17a0:  3fff2d9c 3fff2fc4 3fff2f84 402079d9
3fff17b0:  3fff4024 0000000f 00000001 402057b4
3fff17c0:  3fff2fc4 00001388 3fff0840 00000001
3fff17d0:  00000001 40206b8c 0000000f 3fff0840
3fff17e0:  00000000 00000000 3ffefb08 3fff0830
3fff17f0:  00000001 3fff2fa8 3fff2f84 40207b63
3fff1800:  3ffe96a8 00000000 3fff0840 00000000
3fff1810:  3fffdad0 3fff0830 40202e98 3fff0840
3fff1820:  3fffdad0 00000000 3fff0828 4020cda6
3fff1830:  00000000 00000000 00000001 40217100
3fff1840:  3fffdad0 00000000 3fff0828 40202ee4
3fff1850:  feefeffe feefeffe 3fff0840 40203e2c
<<<stack<<<
```

arendst, 2017-09-05:
Will investigate. pls provide output of command status 0 too.

arendst, 2017-09-05:
Probably due to i18n just implemented. I'll investigate next days.

atbrask, 2017-09-05:
I've also encountered this problem. I agree that it looks like it's related to i18n. When navigating to e.g. the information page, the stack trace shows a crash in line 1480 of webserver.ino which says:
`page.replace(F("{v}"), S_INFORMATION);`
It seems to resolve the issue (for this particular instance of the crash) if I change this line to:
`page.replace(F("{v}"), D_INFORMATION);`

arendst, 2017-09-05:
It did work fine when I tested it but it makes perfectly sense that that's the main cause. If so then there are a few other places thet may fail. Just search for {v} and replace all references from S_ to D_.

When I'm back at 'my office' I will dig deeper.

arendst, 2017-09-07:
Pls give 5.6.1l a try and let me know if it fixed your problem.

jojost1, 2017-09-07:
I think it fixed this particular issue, thanks.

PeggyFree, 2017-09-07:
By the way, even if it's not the best place to say that, the EU version is only 4.92€ (5.80$)  these days, I just ordered 13 units.

c2342, 2017-09-07:
Confirmed. It works now - thanks heaps!

closed 2017-09-07 by c2342