arendst/Tasmota #831: Cross Switch with Sonoff, Possible?

ChrSch84, 2017-09-04:
Hi,

i would like to use 2 Sonoff Touchs for 1 Room. The Problem is that Sonoff 1 (x) dont know the state of sonoff 2 (y). 

With no sonoff i use cross switch for on/off the light.

Is it possible to make device x aware of the state of device y?

I hope you understand what i mean.
Greatings Chris

electrontube, 2017-09-04:
you just need one switch (x) connected to the load (lights or whatever) use the second one (y) simply to toggle (x) via MQTT.
See Wiki ...
SwitchTopic / ButtonTopic 
etc.

lobocobra, 2017-09-04:
**You could also do the following...**
When you have X-switch in place, then you have for sure at least 1 cable between the 2 switches, as this is needed to have x-switch in the traditional way. 
1) attack a thiner 2 cable to this cable and pull it out to the other side.
2) attach to one side the old switch and in the other place you have sonnoff with the switch attached to it to GPOx
=> like this you avoid a dependency to a MQTT server, which brings you in trouble with your wife, if one day your MQTT server is down, because you were optimizing something. 

While I love homematic, I even love the piece at home and thus not to be dependent for light on a working wireless and connection to a MQTT server.

electrontube, 2017-09-04:
Why would one buy wifi controlled switches, flash those with an MQTT enabled firmware then run wires between the switches to bypass wifi and MQTT?

Don't like wireless?
Go for wired components!

ionciubotaru, 2017-09-04:
I assume that sonoff1 controls the lights in your room.
on sonoff2 set `buttontopic sonoff1`

When you touch sonoff1 it will toggle (normal behaviour)
When you touch sonoff2 it will publish `cmnd/sonoff1/power TOGGLE`. After receiving this message sonoff1 will toggle.

ChrSch84, 2017-09-04:
Thank you for all the info.

Drag a cable is not an option :(

@electrontube hmmm sounds good... i have to look at my switches...but i dont know how to switch one sonoff  touch with a other sonoff touch.

I am new at MQTT and the Wiki is a bit hard to understand. I am happy that i could read out the Temp Status and the Humity Status  :D so dont blame me.

davidelang, 2017-09-06:
even better, you can define a group topic for switches and have them all user 
buttontopic set to that topic.

that way, when any of them press a button, it sends the toggle command to the 
mqtt server, and the devices all listen for that and switch.

David Lang

On Mon, 4 Sep 2017, Ion wrote:

> I assume that sonoff1 controls the lights in your room.
> on sonoff2 set `buttontopic sonoff1`
>
> When you touch sonoff1 it will toggle (normal behaviour)
> When you touch sonoff2 it will publish `cmnd/sonoff1/power TOGGLE`. After receiving this message sonoff1 will toggle.

ionciubotaru, 2017-09-06:
@davidelang your solution is fine, till one device miss a toggle command. Then one will be ON and the other will be OFF, and all future TOGGLE commands will keep them in this state.

The best way is to define (in your HA system) a device as master and change the slaves after master's state.

mkh595, 2017-09-06:
Why one sonoff has to know state of other? It is not necessary to be synchronized.

I replaced 3 cross switches with 3 Sonoff basic. Sonoff1 controls light with relay. Others send TOGGLE comand via ButtonTopic for Sonoff1 (not use relays).

**@ChrSch84** Everything you need was mentioned in 2. and 5. post.

**@ionciubotaru** And what happen for your _"best way"_ if WiFi, MQTT or HA will be done?

lobocobra, 2017-09-06:
@ electrontube 
Quite an useless remark from your side, but I will tell you why. 
=> for light you want a system that works for 100% and not 99%
=> you might want the possibility to centrally turn it on/off or automatically out after 120min, but when you are in the cellar in the dark, you want to push a button and you want that it works.
=> you might want not to have 2 devices using 7/24 electricity, just to be used 4x a week
=> you might want to reduce investment and only add new switches, where you really need it
=> ever heard from the demo-effect? "ahh you want to turn on the light, well MQTT is down, don't know why, but push the other button.

With your proposal, all this is not guaranteed as many devices are between your simple action to push the button and the light you want to turn on.
=> there is a reason why even in new houses the switches have additionally wireless connection but do not depend on a central server.

This is why in my house I can turn/on off all switches wireless, but if the Openhab,CCU2,MQTT,Router,Switch or similar is down, I still can turn on any light. The additional of my proposal work is 30 Minutes and the best is, that I need less devices in my house that use electricity 7/24, just to be used 3x / week.

electrontube, 2017-09-06:
@lobocobra 

Have you ever thought about sonoffs hardware / firmware failing or simply a power outage?
Don't forget to add a UPS and some extra / spare switches, have some extra lights bulbs handy, or you will not even get close to 100% availability:-)

@ChrSch84 was looking for a simple solution, he didn't want to control a nuclear power plant

ChrSch84, 2017-09-06:
@all 
Thank you all for the input. If it is possible to connect both switches with a wire i will do it. But this is a very very very tricky electro install. So it is possible that i must use the 1 Main Switch and 1 Remote Switch.

The Problems i have is my time, my (poor) know how of the sonoff/mqtt,and the wife/Child :)
Thats why i ask if it is possible to add this to sonoff FW. Like a Box where you could drop in the mqtt name of the other switch and all is done *dream*.

electrontube, 2017-09-06:
@ChrSch84

if you don't have the resources (time) to install a simple MQTT Broker, than you will either have to modify your cabling, or modify the Sonoff Hardware, or both.
I doubt the Firmware will be modified in the way you requested. I already requested something similar a while ago.
Idea was an option to send status messages via http in order to integrate sonoff/tasmota in non MQTT environments.

Search the web for mosquito, get a cheap Raspberry, (even a "Zero W" would do and you enter the world of mqtt

ionciubotaru, 2017-09-06:
@ChrSch84

Try https://www.cloudmqtt.com/ it's free up to 10 devices. Create your free account, then put the credentials in Tasmota's MQTT configuration web page. There are lots of free mqtt brokers on internet, just open your mind to new technologies and use them.

ChrSch84, 2017-09-06:
@all 
i have a openhab server with a mqtt broker@rpi3 :) but i have to learn how to use it as you told me. I could on/off my devices but thats it. So thanks for the infos but dont flame the mqtt noob.

closed 2018-05-07 by stale[bot]