arendst/Tasmota #832: After the restart of homeassistant, the switch is out of sync.

lovinggrey, 2017-09-04:
If my switch is turned on, the status of the switch is off after restarting the homeasststant. Is there any way to solve it?

ionciubotaru, 2017-09-04:
Upgrade to the latest version - problem was solved

lovinggrey, 2017-09-04:
OK, I'll try. Thank you

mkh595, 2017-09-04:
If your switch doesn't send mqtt message with retain flag then HA is not able after restart show correct value. It is not related to version, I have 0.52.1 with same behaviour like before.

lovinggrey, 2017-09-04:
You mean set up in logging?

mkh595, 2017-09-04:
It is from HA documentation:
_In an ideal scenario, the MQTT device will have a state_topic to publish state changes. If these messages are published with RETAIN flag, the MQTT switch will receive an instant state update after subscription and will start with correct state. Otherwise, the initial state of the switch will be false/off._

HA doesn't remember state of switch after restart, it is updated from first message that comes after restart, message with retain flag comes immediately, without retain flag it dependes on telemetry period.

perozonp, 2018-01-06:
I solved the problem with this automation (where "sonoffs" is a group topic)

- alias: 'Refresh MQTT Switches'
  trigger:
    platform: homeassistant
    event: start
  action:
    service: mqtt.publish
    data:
      topic: cmnd/sonoffs/POWER

closed 2018-05-20 by stale[bot]